Klipper3d/klipper #633: Re-ARM + RepRapDiscount 128x64 Full Graphic Smart Controller problem

J-charles-C, 2018-09-10:
Hello,

I have a Re-arm and a RepRapDiscount 128x64 Full Graphic Smart Controller.

The screen whistles slightly continuously. The sound is weak and does not seem to come from the buzzer but it is painful.

If I comment it in the printer.cfg, it is always lit but no longer updates the info. He still whistles.

If I unplug EXP2, it whistles too.

best
JC
[klippy.log](https://github.com/KevinOConnor/klipper/files/2365936/klippy.log)

J-charles-C, 2018-09-10:
I,

With Marlin, i don't have this problem.

I have check my wire, it seems ok.

I connected the screen like this:
[image]

thanks for your help.

AxMod3DPrint, 2018-09-10:
While it may not have happened with Marlin, it definitely is a hardware issue opposed to a firmware one, that's specific to a Re-Arm. If the LCD does not initialise it will sound the buzzer at full volume, this I have learnt from being deafened while getting the Klipper configs together and making sure the LCD worked (I sorted the Re-Arm generic config), so I do know what you are talking about, but it really is a hardware issue and not a firmware one, it's just the way the Re-Arm handles booting and LCDs that causes this. I don't think there is anything that can be done firmware-wise to fix this, although I'll leave that for @KevinOConnor to have the last say.

CHILLYSMOKES, 2018-09-10:
I had the same problem when trying to activate the LCD on Re-Arm. My BLTouch and LCD would hiss and the LED would fluctuate when I changed the config for the LCD. I came to the conclusion that one of the pins called out for the display was jacking my BLTouch... So I added another MCU, the old MEGA I replaced with the Re-Arm and a spare Ramps. Put the display on that and haven't looked back.

J-charles-C, 2018-09-10:
Ok,

I don't want add a second MCU for one LCD.

If I remove the buzzer the problem would be solved?

best

KevinOConnor, 2018-09-10:
If you add the following to your config, does it cause the noise to stop?
```
[static_digital_output buzzer]
pins: !P1.31
```

If that turns the buzzer full on, try it without the '!'.

-Kevin

J-charles-C, 2018-09-11:
I,
with or without the!, I always have the very slight hissing like a capacitor that vibrates.

Thanks

CHILLYSMOKES, 2018-09-11:
For reference, my LCD worked perfectly before I added BLTouch and SPI pins for TMC2130 support to the config. Do you have either?

J-charles-C, 2018-09-11:
I CHILLYSMOKES,

Unplug BL-touch don't stop the noise.

I don't use 2130 SPI.

thanks

AxMod3DPrint, 2018-09-11:
@J-charles-C while I understand it seems like an issue with klipper, I initially had this issue, so I do know what you are talking about. However, it turned out to be an issue with the way the Re-Arm & LCD themselves interact. In my testing to get the LCD working with the generic Re-Arm config I discovered it is basically your LCD not being initialised properly as when it was initialised properly the hiss disappeared. As a check I swapped out my LCD and cables and the problem went away, which suggested the issue being the LCD over Klipper.

J-charles-C, 2018-09-11:
I KevinOConnor, you're a genius !

[static_digital_output buzzer]
pins: !P1.3**1**

Quiet is great !

Many many thanks

J-charles-C, 2018-09-12:
I,

This is my config for this screen, menu and kill buton works 👍 

`# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
# Re-Arm will only work with this type of display
[display]
lcd_type: st7920
cs_pin: P0.16
sclk_pin: P0.15
sid_pin: P0.18
encoder_pins: ^!P3.25, ^!P3.26
click_pin: ^!P2.11
kill_pin: ^!P1.22
[static_digital_output buzzer]
pins: !P1.30
`

Cordially

KevinOConnor, 2018-09-12:
Thanks - I updated the example config file with this (commit d459ae64).

-Kevin

AxMod3DPrint, 2018-09-12:
Hah, was just in the middle of PR to do the same thing...

KevinOConnor, 2018-09-12:
Oh - just noticed the encoder_pins have an ! in them.  Can you verify that is correct?  (On most boards the ! results in the menu having a delay between rotating forward and rotating backwards.)

-Kevin

J-charles-C, 2018-09-12:
Hello,

It works better without it! It's more fluid.

Sorry, as I do not know much about programming, I tested all possibilities.

I edited the previous post to deceive anyone

best

KevinOConnor, 2018-09-12:
Okay, thanks.  I've updated the example config as well (commit 32b763ef).

closed 2018-09-12 by J-charles-C