Klipper3d/klipper #635: Attempting new Printer configuration- Wanhao Duplicator 6/ Monoprice Ultimate

cliffraccani, 2018-09-10:
As far as I know, no one has produced a printer.cfg for the above printers.

I have attempted to do so with the following results:

Thinkpad X200s running Raspian Desktop.
DEBIAN STRETCH WITH RASPBERRY PI DESKTOP
The Raspberry Pi Desktop OS for PC and Mac - based on Debian Stretch
Version:June 2018
Release date:2018-06-27
Kernel version:4.9

Octoprint has worked fine on this setup without Klipper.

The printer is directly connected via USB to this Laptop which is connected to my router via Ethernet cable.
I installed Klipper and all went well.

After creating a printer.cfg file from Marlin to get pin assignments etc (see .zip attachment). I ran through the configs check document- Verify temperature, Verify M112, and so on up to Verify Stepper Motors. Did the STEPPER_BUZ command for x, y, z and looked good- going in right direction and completed 10 movements.

I next did a G28 X and the stepper just buzzed and shook and and did not home. Same for y, z steppers.
I have not got any further than this.

Can't get past G28- homing. All pin settings seem OK according to configuration tests. End stops trigger on all axis. Steppers move in right direction with STEPPER_BUZ. What would cause the steppers to just vibrate and buzz and not move. Speeds, acceleration settings. Klippy.log gives no clue. 

Any ideas?
Thank you for your time.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2367489/klippy.log)
[Klipper2MarlinNotes.zip](https://github.com/KevinOConnor/klipper/files/2367929/Klipper2MarlinNotes.zip)

KevinOConnor, 2018-09-10:
It sounds a bit like: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-converted-my-config-from-marlin-and-the-xy-axes-work-fine-but-i-just-get-a-screeching-noise-when-homing-the-z-axis
So, you should try with a much lower homing speed.

Other than that, the symptoms also seem similar to a blown stepper driver or an incorrect wiring between stepper motor and stepper driver.

-Kevin

cliffraccani, 2018-09-11:
Thanks for the reply. 
Definitely not blown drivers or incorrect wiring!
The motors do not have much torque it seems.
Would you know how to translate this Marlin code to Klipper:
from ..Marlin/pins_ULTIMAIN_2.h file,

#define MOTOR_CURRENT_PWM_XY_PIN 44
#define MOTOR_CURRENT_PWM_Z_PIN 45
#define MOTOR_CURRENT_PWM_E_PIN 46
// Motor current PWM conversion, PWM value = MotorCurrentSetting * 255 / range
#ifndef MOTOR_CURRENT_PWM_RANGE
  #define MOTOR_CURRENT_PWM_RANGE 2000
#endif
#define DEFAULT_PWM_MOTOR_CURRENT  {1300, 1300, 1250}

Many thanks.

cliffraccani, 2018-09-11:
OK made some progress. converting above to Klipper:

[output_pin stepper_xy_current]
pin: ar44
pwm: True
scale: 2.0
cycle_time: .002
hardware_pwm: True
static_value: 1.3

[output_pin stepper_z_current]
pin: ar45
pwm: True
scale: 2.0
cycle_time: .002
hardware_pwm: True
static_value: 1.3

[output_pin stepper_e_current]
pin: ar46
pwm: True
scale: 2.0
cycle_time: .002
hardware_pwm: True
static_value: 1.25

Does this look correct?

KevinOConnor, 2018-09-11:
Yes, if the board is setting up driver current via a pwm pin, then the above looks correct.  What micro-controller board are you using?  If it's a "mini rambo" then you should use config/generic-mini-rambo.cfg as a starting point (that config also configures the driver micro-stepping pins).

-Kevin

cliffraccani, 2018-09-11:
The board is a Wanhao Duplicator 6, which is a clone of the Ultimaker 2 motherboard. I have not found any micro stepping pins definitions in the code as yet. The movement is looking much better now, like a 3d printer. There is a high pitched whine coming from the motors now.  Always somthing else.

cliffraccani, 2018-09-11:
Thank you Kevin.
You are a star.
That cycle_time did it, I don't know how you came up with that figure, I'd like to know how. 
I've learnt a lot over the last few days. From a total noob to just a noob!
I could not find any MS pins to set 1/16 micro stepping in the Marlin firmware for this board. Do know how I could find out? Maybe the drivers are setup in hardware, jumpers, to give the highest level of micro stepping. I found the mini_rambo board MS pin definitions but none for the UltiMain_2 board.

The printer is working great right now. I am printing at 100mm/sec and I've never seen this printer do that without under extrusion.

Pressure advance next.
Then see if Klipper can be configured for the UltiLcd_2 display and dial. I would be happy with a simple display of printer progress and speed, layer, temperature etc info.

I would like to make this basic printer.cfg file available so others can improve on it.

cliffraccani, 2018-09-11:
Scratch that idea of using the existing display. Not worth the bother. I have an old smart mobile phone with a great touchscreen, that should be perfect. Print a mount for it in front of old screen and job done. Got used to the old method of controlling printer but did I really lie it? Not really. 
Loving this setup. This all works off a 16GB USB Flash drive with live Raspian desktop. Just plug into what ever computer is handy. I have many old laptops lying around as many people seem to have many RPi's.
Am going to do a pull request to add this printer.cfg to the list of printers supported and take it from there. 
Thanks again for your help. No doubt I will be back with more questions.

KevinOConnor, 2018-09-20:
Looks like this discussion has moved to pull request #648, so closing this one.

-Kevin

closed 2018-09-20 by KevinOConnor