Klipper3d/klipper #636: Discussion: Shutting down MCU part (printer itself)

mcmatrix, 2018-09-11:
I'm shutting down the printer when I'm not using it. The RPi remains running.
Now I have SSR connected to RPi gpio17. This relay is controlling
printer power. Previously I was manually turning on/off printer power.

When using single firmware like Marlin then everything was simple.
When the printer was off then it was offline in Repetier-Server, when I turned power on
it went green (online) in the server.

When using Klipper then it's all the time online (from the server point of view) even when the MCU
part is down. But internally Klipper is in "error" state?

> As in Klipper,  the micro-controller software and host software are
> conceptually a "single firmware" - so it doesn't really make sense to
> externally remove power to "half the printer".

So when I turn on printer power Klipper itself doesnt do anything
it remains in "error" state.
I have to issue `FIRMWARE_RESTART` gcode to get Klipper back to the normal state.

> If there is interest in extending the code to better support this, I
> would recommend enhancing Klipper so that it knows how to remove power
> to the micro-controller on idle.  Then, it can also be enhanced to
> reconnect to the micro-controller when the printer is no longer idle.

Maybe there can be done something internally to make Klipper handle
it better.

I'm starting with Klipper and trying to get it running the way I like and have used to be.

Guys, what are your tricks and procedures when turning on/off the printer?

CHILLYSMOKES, 2018-09-11:
#581 , #546

mcmatrix, 2018-09-11:
OK, found firmware (Klipper) related XML config file from Repetier-Server.
I'll try to play with these settings and make Repetier better understand the Klipper

```xml
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<firmware>
    <name>Klipper</name>
    <connectionCommands>M110 N1
        M105
        M115
        M220 S100
        M221 S100
        G92 E0
        G90
        M82
        G21
        @getip
    </connectionCommands>
    <com okAfterResend="false" protocol="ascii" allowColonInAscii="false"/>
    <eepromType></eepromType>
    <temperatureOutputMultiplier>2</temperatureOutputMultiplier>
    <busyHandling>relaxed</busyHandling>
    <blockingCommands>
        <command>M109</command>
        <command>M190</command>
        <command>M116</command>
        <command>M303</command>
        <command>M400</command>
        <command>M226</command>
        <command>M600</command>
        <command>M0</command>
        <command>M1</command>
        <command>M48</command>
        <command>M300</command>
        <command>M862</command>
        <command>M863</command>
        <command>G4</command>
        <command>G28</command>
        <command>G29</command>
        <command>G32</command>
        <command>G30</command>
        <command>G33</command>
    </blockingCommands>
    <gcodeWithText>
        <m>20</m>
        <m>23</m>
        <m>28</m>
        <m>29</m>
        <m>32</m>
        <m>30</m>
        <m>36</m>
        <m>117</m>
        <m>3117</m>
        <m>531</m>
    </gcodeWithText>
    <priorityCommands>
        <command>M112</command>
    </priorityCommands>

    <command type="setFlowMultiply">M221 S{i:flowPercent}</command>
    <command type="setSpeedMultiply">M220 S{i:speedPercent}</command>
    <command type="requestTemperature">M105</command>
    <command type="disableFan">M107 P0</command>
    <command type="setFanSpeed">M106 S{i:fanSpeed} P0</command>
    <command type="disableFan2">M107 P1</command>
    <command type="setFan2Speed">M106 S{i:fan2Speed} P1</command>
    <command type="motorsOff">M84</command>
    <command type="setBedTemperature">M140 S{i:temperature}</command>
    <command type="setChamberTemperature">M141 S{i:temperature}</command>
    <command type="setChamberTemperatureWait">M191 S{i:temperature}</command>
[... 93 lines cut ...]
    <setting type="XMin">^echo:Soft.*Min:.*X([+-]?\d+\.?\d*)</setting>
    <setting type="YMin">^echo:Soft.*Min:.*Y([+-]?\d+\.?\d*)</setting>
    <setting type="ZMin">^echo:Soft.*Min:.*Z([+-]?\d+\.?\d*)</setting>
    <setting type="XMax">^echo:Soft.*Max:.*X([+-]?\d+\.?\d*)</setting>
    <setting type="YMax">^echo:Soft.*Max:.*Y([+-]?\d+\.?\d*)</setting>
    <setting type="ZMax">^echo:Soft.*Max:.*Z([+-]?\d+\.?\d*)</setting>

    <!-- The following parameter can not be obtained from marlin
        <setting type="extrJerk" indexAdd="-1" index=" Extr.(\d+) ">(\d+\.?\d*) Extr.\d+ start feedrate \[mm/s\]</setting>
        <setting type="XMin">([+-]?\d+\.?\d*) X home pos \[mm\]</setting>
        <setting type="YMin">([+-]?\d+\.?\d*) Y home pos \[mm\]</setting>
        <setting type="ZMin">([+-]?\d+\.?\d*) Z home pos \[mm\]</setting>
        <setting type="XLength">(\d+\.?\d*) X max length \[mm\]</setting>
        <setting type="YLength">(\d+\.?\d*) Y max length \[mm\]</setting>
        <setting type="ZLength">(\d+\.?\d*) Z max length \[mm\]</setting>
        <setting type="baudrate">(\d+) Baudrate</setting>
        <setting type="xyJerk">(\d+\.?\d*) Max. jerk</setting>
        <setting type="zJerk">(\d+\.?\d*) Max. Z-jerk</setting>
        <setting type="printableRadius">(\d+\.?\d*) Max printable radius</setting>
        <setting type="printerType" value="1">Z\-axis acceleration</setting>
        <setting type="printerType" value="2">Diagonal rod length</setting>
        <setting type="sdInstalled" value="1">^SD printing byte</setting>
        <setting type="fanInstalled" value="1">^Fanspeed:</setting> -->
</firmware>
```