Klipper3d/klipper #638: (mesh bed leveling) : spiralize the grid ?

aeropic, 2018-09-11:
I just would want to modify the bed_mesh.py in order to follow a spiral pattern from the center of the bed plate instead of the zig zag one. 
(As the homing point, in my case, is the center of the bed, I want to suppress an initial move to the corner and also to finish by the front left corner of the bed to allow better access to the head for a cleaning at that location...)

It seems this could be a very simple mod, but before doing it I would want to be confirmed that the order the probe locations are stored into the "points" array does not matter ? I mean can I store my points in spiral rather than in zig zag with the points.append((posx, posy)) instruction?

Thanks

points = []
        for i in range(y_cnt):
            for j in range(x_cnt):
                if not i % 2:
                    # move in positive directon
                    pos_x = min_x + j * x_dist
                else:
                    # move in negative direction
                    pos_x = max_x - j * x_dist
                points.append((pos_x, pos_y))

Arksine, 2018-09-11:
The way in which the points are stored matters.  The finalize() function expects the points to return in a zig-zag fashion so that it can correctly organize the table of z-values used to generate the mesh.   This is why I chose to procedurally generate the points, so that the user didn't enter invalid points that would confuse mesh generation.

aeropic, 2018-09-11:
Thanks Eric, I'll see if it still worths the extra effort ;-)

aeropic, 2018-09-11:
@Arksine , well, I tried but I must say I'm stuck very quickly ;-(
(did I mention I am a fully noob in python ?)

I cannot even add one simple line in your code, just trying to insert a log line stalls everything 
eg : replacing this line
logging.info('bed_mesh:  generated points')
by this one is OK
logging.info('bed_mesh: my generated points')

but duplicating it does not work...
logging.info('bed_mesh: my generated points')
logging.info('bed_mesh: my generated points')
==> It compiles but it does not run anymore...  

 What do I do wrong ? I admire you, fluent people in python !!!!

Arksine, 2018-09-12:
@aeropic Most likely its a formatting issue.  For example, if your editor is inserting tabs instead of spaces for indentation the interpreter won't like mixing them.

aeropic, 2018-09-12:
Bingo, the editor replaced the 4 spaces by one tab! And I've now learnt my first lesson of Python ;-)
Thanks, now the fun will begin ... I keep you informed!

aeropic, 2018-09-17:
Just for the fun , I eventually did it which helped me to better understand the mesh levelling code ;-).
- I added a "grid_pattern: spiral" (or zigzag by default) in the printer.cfg. It is only effective if the grid is squared with the same  x_cnt and y_cnt and if the x_cnt is a odd number
- I modified both modules _generate_points(self, config) and finalize(self, offsets, positions):
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2390214/printer.cfg.txt)
[bed_mesh.py.txt](https://github.com/KevinOConnor/klipper/files/2390216/bed_mesh.py.txt)

It is working like a charm with, to my eyes, the advantages to both minimize the traject of the probe and exit at the point (0,0) which is easier to clean the noozle on the fly ...

Fell free to keep it and merge it in the product if you thibnk it worths. (Mind the G28 line which is commented)
#self.gcode.run_script_from_command("G28")

Thanks again for the great work done here.
Alain

aeropic, 2018-10-11:
As this post is referenced in #681, I suggest we can close it !

closed 2018-10-11 by aeropic