Klipper3d/klipper #640: FW communication error, requires fw restart when heating bed and delta calibrating

ysoldak, 2018-09-11:
I'm calibrating my delta manually (with paper and NEXT command) and with bed heated to 60, to eliminate any error for heat expansion (if any).

Seems like heating bed and doing heavy calibration calculations do not go together and the system errors out requiring `FIRMWARE_RESTART` directly after printing out calibration results. Seen this before, thought that was a fluke, but it does happen very often, so posting the log here.

Will try disable bed heating directly before last NEXT and see what happens, will report here.

[klippy_fw_error_after_delta_calibrate.log](https://github.com/KevinOConnor/klipper/files/2372358/klippy_fw_error_after_delta_calibrate.log)

KevinOConnor, 2018-09-12:
This looks like Python GIL madness.  I'll have to think about how to fix this.

FWIW, I always measure with the extruder and bed at room temperature.  I've found that the heat expansion is predictable, so I just account for it separately.  (Specifically, to account for expansion, I'll print a test object like the square in docs/prints/square.stl with a coarse layer height and then look for a squashed first layer - making z adjustments until the first layer matches the other layers.)

-Kevin

ysoldak, 2018-09-12:
Re-trying basic calibration on the same branch. No bed was heated this time, so all went OK until I've tried manually paper-check the points after calibration, once I command to move to (-43.3,25,5) point (this is "far b" so to say) it hangs. Repeatable, 2 times in a row, second time directly after `restart` and `firmware_restart` and no calibration after. Calibration config was saved and is in "printer.cfg".
See log attached. 

[klippy_fw_error_far_b.log](https://github.com/KevinOConnor/klipper/files/2377068/klippy_fw_error_far_b.log)

KevinOConnor, 2018-09-13:
The "klippy_fw_error_far_b" error is a general Klipper error (not related to the new delta calibration code).  It should now be fixed on both master and the delta calibration branch (commit 390d549c).

Thanks,
-Kevin

KevinOConnor, 2018-10-15:
I believe this issue should now be fixed (commit 368703fd).

-Kevin

ysoldak, 2018-10-16:
Thanks, I'll verify shortly. Waiting for a delivery to re-assemble my bot.

ysoldak, 2018-10-18:
I can confirm latest master works for me (pulled on 2018-10-17). DELTA_CALIBRATE command succeed without hanging, while it was 100% handing before. Can close the issue, i suppose.

closed 2018-10-22 by KevinOConnor