Klipper3d/klipper #643: Menu PSU Control

SergeyKrutyko, 2018-09-12:
I have a klipper with menu features. I make menu items for PSU Control Plugin:

....
[menu control_pon]
type: command
name: Power ON
gcode: M80

[menu control_poff]
type: command
name: Power OFF
gcode: M81
....

If i select this command from menu then get:
Recv: // Unknown command:"M81"
Recv: // Unknown command:"M80"

If i write it from terminal then all ok:

Send: M80
PSUControl: ok

and it worked ok....

Please help me!

mcmatrix, 2018-09-12:
The Klipper itself doesn't support M81, M80. 
Therefore Klipper reports that it's `Unknown command`.
Where is your PSU on/off pin connected (Raspberry or printer controller)?

SergeyKrutyko, 2018-09-13:
i am sorry, but my systems have a control of ATX power via pin of Orange PI......  

There is no other solution?

AxMod3DPrint, 2018-09-13:
If you are using the PSU Control plugin, which it looks like you are, it _may_ be possible to hook into that. However, it would probably need digging around in the code for that plugin. @mcmatrix will be able to confirm if this is possible or not.

mcmatrix, 2018-09-13:
@SergeyKrutyko Could you post screenshot of your PSU Control settings.

AxMod3DPrint, 2018-09-13:
Two switching methods: GCode Command and System Command.

[image]
[image]

SergeyKrutyko, 2018-09-13:
[image]

SergeyKrutyko, 2018-09-13:
@AxMod3DPrint the problem is that when using the GCODE in PSU Plugin works from the terminal, but does not work from the menu

mcmatrix, 2018-09-13:
@SergeyKrutyko Maybe you could try **Action Commands** plugin
https://plugins.octoprint.org/plugins/actioncommands/

In Octoprint you have to define 2 additional actions
`poweroff` _type:system_ `gpio mode7 output & gpio write 7 0`
`poweron ` _type:system_ `gpio mode7 output & gpio write 7 1`

And from menu you have to send action to Octoprint
```ini
[menu control_pon]
type: command
name: Power ON
action: respond action:poweron
gcode:

[menu control_poff]
type: command
name: Power OFF
action: respond action:poweroff
gcode:
```

AxMod3DPrint, 2018-09-13:
@SergeyKrutyko realise that. Settings screens were there as an example.

SergeyKrutyko, 2018-09-13:
@mcmatrix NOT WORK VIA ACTION..... log file nothing too..... (((

CHILLYSMOKES, 2018-09-13:
@mcmatrix Can you just make another menu 'element' that will execute system commands? You have already called os. in your header.  This will remove the need to interface Octoprint.  Lemme know if I'm close. 8)

mcmatrix, 2018-09-14:
@SergeyKrutyko What didn't work? To help you I need more info.
This is how I understand the PSU plugin is working. Correct me if I'm wrong.
You have defined 2 On System commands to turn gpio on/off.
Octoprint is monitoring gcodes that are feed thru it and will catch `M80`, `M81`.
When seeing `M80` it will execute **ON command**, when `M81` then **OFF command**.

**The Klipper won't see these because Octoprint will catch these gcodes and these will not pass. 
Klipper doesn't support such gcodes.** 

To make it work you have to use additional plugin `Action Commands`. 

Did you defined two actions `poweroff`, `poweron` in that Octoprint plugin
and made changes in menu items?
When you execute menuitem what you'll see in Octoprint console?

For example when executing `Power ON` from menu it should send
`//action:poweron` to the host (octoprint). Octoprint plugin is monitoring these and will
execute whatever commands you have attached to these actions.

mcmatrix, 2018-09-14:
> @mcmatrix Can you just make another menu 'element' that will execute system commands? You have already called os. in your header. This will remove the need to interface Octoprint. Lemme know if I'm close. 8)

@KevinOConnor What you think about this? I can add additional function to `action:`. Lets say `call <whatever you want to call from shell>`. Is there a safe way to execute shell commands from Klipper (sync, async) without breaking Klipper workflow?

KevinOConnor, 2018-09-14:
> @KevinOConnor What do you think about this? I can add additional function to action:. Let's say call <whatever you want to call from shell>. Is there a safe way to execute shell commands from Klipper (sync or async?) without breaking Klipper workflow?

I'd prefer to avoid that.  (Running shell commands can lead to flow control problems and security concerns.)  I think the long term solution would be to enhance the Klipper Linux MCU code to support toggling gpio pins on the Raspberry Pi.  Then flipping the power pin would be just like flipping any other pin in Klipper.

-Kevin

galosre, 2020-03-07:
@mcmatrix  Sorry to reopen this. But i am anxious to get it working with PSU Control when switching Method is "Gpio pin#" what should be the fields in action commands?
Janar Sööt
[image]

ld3300, 2020-03-07:
Hi @galosre ,

This is how I have implemented it:
I actually use the gcode response method, but it should work either way (as long as you have respond declared in your config file

```
[respond]
default_type: echo
```

```
[menu __klipper __shutdownnow]
type: command
name: Shutdown Now
enable: true
gcode: RESPOND TYPE=command MSG=action:M81
```
Then in the Action Commands Octoprint plugin (not action commands prompt) setup something like this (the first line is what you would want):
[image]

Then in PSU control set this up:
[image]

and for good measure in Serial Connection under Firmware and Protocol I block those commands from being sent to the printer (just so it doesn't complain at me):
[image]

galosre, 2020-03-07:
I got it working for a while with M81 but I wanted to try your other suggestion from the other thread: remove the "action:" from the last line and try it. so it just says
gcode: RESPOND TYPE=command MSG=poweroff but this definetly was not working but now with M81 got just  `19:17:16 action:M81` This is very strange!

And about the same topic how can I trigger this Gcode macro?
`gcode: RESPOND TYPE=command MSG=poweron` does not bring me any joy!
[image]

ld3300, 2020-03-07:
my bad, the action:M81 is supposed to be there.  I was looking at the wrong thing.

Assuming your host processor isn't powered from your psu then startup would be the same, but you would need another action command for the firmware restart.  If your host is powered from the PSU (as mine is), then you cannot send an action command to turn it on.  You have to configure the power on with the top bar button, or by one of the other means PSU control documents.

you can't put gcode macros in action commands like that (gcode: RESPOND TYPE=command MSG=poweron), they only exist in klipper.  

I essentially wrote the tutorial for you a couple of posts ago and you got it working.  You just need to keep applying the same principles to execute other Action Commands.  The principles are you send a RESPOND TYPE=command MSG=action:(whatever)
Put the (whatever) in the action field
put the gcode or name of a klipper gcode_macro you want to spit out in the last field (ie M80, START_PRINT, firmware_restart, whatever), and when the action command is received the value will be sent.

Spend some time looking through the klipper supported gcodes file and the gcode macros section of the example-extras.cfg file to learn what can be implemented.

galosre, 2020-03-07:
Finally the menu turn off is working again I think the plugin got confused with the other actions
No, my PI is always on I just want to send firmware _restart, When I click connect in klipper's plugin but this does not appear to be picked up. 
[image]

ld3300, 2020-03-07:
That probably won't work with action commands, but did you try putting it in the octoprint gcode settings screen where it says "after connection to printer is established"

galosre, 2020-03-07:
Thank you so much this is working perfectly
[image]
One last question not related at the end of a print" I got Too many consecutive timeouts, printer still connected and alive?" and this happened before Psu control Trigger Off.

ld3300, 2020-03-08:
Do you have any Kipper macros running at the end of the print that have long wait times?

galosre, 2020-03-08:
Maybe G4 ; Wait
 in cura Custom End G-code
Anyway not a big deal.

galosre, 2020-03-08:
I just realized just adding M80 and M81 in those fields I can start and end (if necessary ) the printer just by clicking connect /disconnect in octoklipper without ever using the green lightning icon from psu control so in this case Action Command is just useful for the menu poweroff
Again Thank you
[image]

whoim2, 2020-05-05:
> Я только что понял, что просто добавив M80 и M81 в эти поля, я могу запускать и завершать (при необходимости) принтер, просто нажимая кнопку «Подключить / отключить» в octoklipper, даже не используя зеленый значок молнии из элемента управления psu, поэтому в этом случае команда Action просто полезна для Poweroff меню
> снова Спасибо
> [image]

Hello! I need delay between board power on and 'firmware_restart', no work as you wrote. Work if type mannualy in terminal..