Klipper3d/klipper #645: CoreXZ new kinematics

malgangda, 2018-09-13:
Hello, I would like to ask if there is the possibility to provide support to another kinematics (CoreXZ)

I am currently building a 3D printer with an endless belt. Similar to the Blackbelt 3D printer.
For this printer I need a COREXZ kinematics in Klipper.
I have already run Klipper on a standard printer and am thrilled.
Hence this request.

What do you think? It is a feasible support?

malgangda, 2018-09-13:
# Code for handling the kinematics of corexz robots
#
# Copyright (C) 2017-2018  Kevin O'Connor <kevin@koconnor.net>
#
# This file may be distributed under the terms of the GNU GPLv3 license.
import logging, math
import stepper, homing

class CoreXZKinematics:
    def __init__(self, toolhead, config):
        # Setup axis rails
        self.rails = [ stepper.PrinterRail(config.getsection('stepper_x')),
                       stepper.PrinterRail(config.getsection('stepper_y')),
                       stepper.PrinterRail(config.getsection('stepper_z')) ]
        self.rails[0].add_to_endstop(self.rails[2].get_endstops()[0][0])
        self.rails[2].add_to_endstop(self.rails[0].get_endstops()[0][0])
        self.rails[0].setup_itersolve('corexz_stepper_alloc', '+')
        self.rails[1].setup_itersolve('cartesian_stepper_alloc', 'y')
        self.rails[2].setup_itersolve('corexz_stepper_alloc', '-')
        # Setup boundary checks
        max_velocity, max_accel = toolhead.get_max_velocity()
        self.max_y_velocity = config.getfloat(
            'max_y_velocity', max_velocity, above=0., maxval=max_velocity)
        self.max_y_accel = config.getfloat(
            'max_y_accel', max_accel, above=0., maxval=max_accel)
        self.need_motor_enable = True
        self.limits = [(1.0, -1.0)] * 3
        # Setup stepper max halt velocity
        max_halt_velocity = toolhead.get_max_axis_halt()
        max_xz_halt_velocity = max_halt_velocity * math.sqrt(2.)
        self.rails[0].set_max_jerk(max_xz_halt_velocity, max_accel)
        self.rails[1].set_max_jerk(
            min(max_halt_velocity, self.max_y_velocity), self.max_y_accel)
        self.rails[2].set_max_jerk(max_xz_halt_velocity, max_accel)
    def get_steppers(self, flags=""):
        if flags == "y":
            return self.rails[1].get_steppers()
        return [s for rail in self.rails for s in rail.get_steppers()]
    def calc_position(self):
        pos = [rail.get_commanded_position() for rail in self.rails]
        return [0.5 * (pos[0] + pos[2]), 0.5 * (pos[0] - pos[2]), pos[1]]
    def set_position(self, newpos, homing_axes):
        for i, rail in enumerate(self.rails):
            rail.set_position(newpos)
            if i in homing_axes:
                self.limits[i] = rail.get_range()
    def home(self, homing_state):
        # Each axis is homed independently and in order
        for axis in homing_state.get_axes():
            rail = self.rails[axis]
            # Determine moves
            position_min, position_max = rail.get_range()
            hi = rail.get_homing_info()
            if hi.positive_dir:
                pos = hi.position_endstop - 1.5*(
                    hi.position_endstop - position_min)
                rpos = hi.position_endstop - hi.retract_dist
                r2pos = rpos - hi.retract_dist
            else:
                pos = hi.position_endstop + 1.5*(
                    position_max - hi.position_endstop)
                rpos = hi.position_endstop + hi.retract_dist
                r2pos = rpos + hi.retract_dist
            # Initial homing
            homing_speed = hi.speed
            if axis == 1:
                homing_speed = min(homing_speed, self.max_y_velocity)
            homepos = [None, None, None, None]
            homepos[axis] = hi.position_endstop
            coord = [None, None, None, None]
[... 41 lines cut ...]
        if (xpos < limits[0][0] or xpos > limits[0][2]
            or zpos < limits[2][0] or ypos > limits[2][2]):
            self._check_endstops(move)
        if not move.axes_d[1]:
            # Normal XZ move - use defaults
            return
        # Move with Y - update velocity and accel for slower Y axis
        self._check_endstops(move)
        y_ratio = move.move_d / abs(move.axes_d[1])
        move.limit_speed(
            self.max_y_velocity * y_ratio, self.max_y_accel * y_ratio)
    def move(self, print_time, move):
        if self.need_motor_enable:
            self._check_motor_enable(print_time, move)
        axes_d = move.axes_d
        cmove = move.cmove
        rail_x, rail_y, rail_z = self.rails
        if axes_d[0] or axes_d[2]:
            rail_x.step_itersolve(cmove)
            rail_z.step_itersolve(cmove)
        if axes_d[1]:
            rail_y.step_itersolve(cmove)

def load_kinematics(toolhead, config):
    return CoreXZKinematics(toolhead, config)

malgangda, 2018-09-13:
// CoreXZ kinematics stepper pulse time generation
//
// Copyright (C) 2018  Kevin O'Connor <kevin@koconnor.net>
//
// This file may be distributed under the terms of the GNU GPLv3 license.

#include <stdlib.h> // malloc
#include <string.h> // memset
#include "compiler.h" // __visible
#include "itersolve.h" // struct stepper_kinematics

static double
corexz_stepper_plus_calc_position(struct stepper_kinematics *sk, struct move *m
                                  , double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    return c.x + c.z;
}

static double
corexz_stepper_minus_calc_position(struct stepper_kinematics *sk, struct move *m
                                   , double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    return c.x - c.z;
}

struct stepper_kinematics * __visible
corexz_stepper_alloc(char type)
{
    struct stepper_kinematics *sk = malloc(sizeof(*sk));
    memset(sk, 0, sizeof(*sk));
    if (type == '+')
        sk->calc_position = corexz_stepper_plus_calc_position;
    else if (type == '-')
        sk->calc_position = corexz_stepper_minus_calc_position;
    return sk;
}

malgangda, 2018-09-13:
# This file serves as documentation for config parameters of corexz
# style printers. One may copy and edit this file to configure a new
# corexy printer. Only parameters unique to corexz printers are
# described here - see the "example.cfg" file for description of
# common config parameters.

# DO NOT COPY THIS FILE WITHOUT CAREFULLY READING AND UPDATING IT
# FIRST. Incorrectly configured parameters may cause damage.

# The stepper_x section is used to describe the X axis as well as the
# stepper controlling the X+Z movement.
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .00623
endstop_pin: ^ar3
#endstop_pin: ^ar2
position_endstop: 0
position_max: 240
homing_speed: 50
homing_retract_dist: 5.0

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .00623
endstop_pin: ^ar14
#endstop_pin: ^ar15
position_endstop: 0
position_max: 350
homing_speed: 50

# The stepper_Z section is used to describe the Z axis as well as the
# stepper controlling the X-Z movement.
[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .00623
endstop_pin: ^ar18
#endstop_pin: ^ar19
position_endstop: 0
position_max: 350
homing_speed: 50

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 5
max_temp: 300

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: watermark
min_temp: 0
max_temp: 130

[fan]
pin: ar9

[mcu]
serial: /dev/ttyUSB0
baud: 115200
pin_map: arduino

[printer]
kinematics: corexz
#   This option must be "corexz" for corexz printers.
max_velocity: 300
max_accel: 3000
max_y_velocity: 25
max_y_accel: 30

KevinOConnor, 2018-09-14:
It's possible to add support for CoreXZ.  See the https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md file for information on adding new kinematics.

If you're looking to contribute new code, you'll need to use git to commit the changes and open a github pull request with those changes.  There's online guides describing that process.  Also see: https://github.com/KevinOConnor/klipper/blob/master/docs/CONTRIBUTING.md .

-Kevin

KevinOConnor, 2020-07-30:
FYI, support for corexz was added as part of PR #3129 .

-Kevin