Klipper3d/klipper #652: Design Enhancement - New Icons Suggestion

alfsoft, 2018-09-15:
Hello. Here are icons I made specially for Klipper. I got tired of standard "FR" icon so I decided to remake it into something understandable visually and language independent. Then I did remake all other icons to match this new feedrate icon. That's how they look:

[image]

If you like them I will make a pull request for icons.py file containing all these icons.

J-charles-C, 2018-09-16:
I,

Nice icon.

It would also be necessary to create zones to separate the infos. It would be more legible.

best

ozligia, 2018-09-16:
+1 vote for new icons

mcmatrix, 2018-09-17:
Same here, nozzle, fan, and bed icons are fine. Also not sure about the feed rate icon.
Should we keep similar icon styles for both hd44780 and graphical LCDs?

I don't know if it's a discussion for this issue but can we do this?
HD44780 custom chars (icons) are the same size as the font so that icons can be used as a character.
Can this be simulated in graphical LCDs? Icons or sprites same size as the font. 
Then you can insert icons into the text without the need to specify exact x,y position.
Example custom screen (card) in HD44780. 
I can easily mix custom chars (icons) and normal charactes.
```ini
### special characters for hd44780
# \x00 - thermometer
# \x01 - heater_bed
# \x02 - speed_factor
# \x03 - clock
# \x04 - degrees
# \x05 - usb
# \x06 - sdcard
# \x7e - right arrow

[menu __card2004_static]
type: card
name: Card 20x04
content:
    "\x00{0:3s}\x04\x7e{1:3s}\x04   {2}"
    "\x01{3:3s}\x04\x7e{4:3s}\x04   \xa5{8}"
    "fn{6} \x02{5} {7}"
    "{9}"
items:
```
It would be wonderful if we can do the same with graphicals LCDs.

marcio-ao, 2018-09-17:
@KevinOConnor: I think these are cool! I'll have to find out whether there is interest in moving this back to our Marlin branch of the UI :)

Arksine, 2018-09-18:
I like these.   Looking forward to seeing what feed rate variants @alfsoft comes up with.

KevinOConnor, 2018-09-18:
> I don't know if it's a discussion for this issue but can we do this?
HD44780 custom chars (icons) are the same size as the font so that icons can be used as a character.
Can this be simulated in graphical LCDs? Icons or sprites same size as the font.
Then you can insert icons into the text without the need to specify exact x,y position.

We can simulate that, but I don't think we want to use special characters in write_text() - instead, we could add a new method like write_glyph(col, row, "hotbed") that would display the appropriate icon.

> HD44780 can only have 8 custom characters simultaneously, these are uploaded to display ram.
So no space for fancy dynamic icons.

FYI, if someone really wanted to, I think it would be possible to do animations on the HD44780.  Although there are only 8 custom glyphs available, it should be possible to change the glyph definition each update to animate that glyph.

-Kevin

mcmatrix, 2018-09-18:
> FYI, if someone really wanted to, I think it would be possible to do animations on the HD44780. Although there are only 8 custom glyphs available, it should be possible to change the glyph definition each update to animate that glyph.

Thanks, good to know that it's possible.

> EDIT: Ah - your question was about inserting icons into strings. We could add a helper function - something like write_icons_and_text("msg and ~hotbed~") - and have it translate to the appropriate low-level write_text() and write_glyph() methods..

That would be super. If we can have that high-level write function for GLCD then it's possible to make nice text screens with icons as HD44780 has. 
Something like this.
```ini
...
name: GLCD 16x04
content:
    "~hotend~{0:3s}~degrees~~rarraw~{1:3s}~degrees~   {2}"
    "~hotbed~{3:3s}~degrees~~rarraw~{4:3s}~degrees~   {8}"
    "~fan~{6} ~speed~{5} {7}"
    "{9}"
items:
...

```
Then we have nice screens for both HD44780 and GLCD.
Btw if you have HD44780 then you can try `__screen2004_static` deck.

marcio-ao, 2018-09-19:
@KevinOConnor: When using the graphics LCD, it is possible to take advantage of the fact that the bitmap buffer is XORed with the text buffer. So suppose you wanted to have two frames of a fan animation, and wanted to update it fast using a single character write, without having to use two of the four custom-character glyphs slots. One way to do this is to paint the first frame of the animation on the bitmap buffer at startup, then generate a custom character glyph which is the XOR of the first and second frame. Now, you can animate the fan simply by alternatively writing the custom-glyph and a space to the text buffer. So, you can actually animate two frames with one custom-glyph.

KevinOConnor, 2018-09-20:
@marcio-ao - that is a neat hack!

Just to be clear though, I don't think Klipper needs to optimize the animations.  There should be enough bandwidth and processing power to write out everything in pure graphics mode - even all the screen text could be written in graphics mode if desired.  (Klipper prioritizes all other activities over lcd updates, so even if there was a resource contention it should only result in slight pauses to screen  updates.)

-Kevin

alfsoft, 2018-09-20:
Sorry for the delay. Here are some of Feedrate icon variants:
[image]

And here are HD44780 icons. Top row is what we have now. Bottom two rows are my variant(s):
[image]

Will try to test these 5x8 icons on real HD44780 + Arduino board as soon as possible.

oderwat, 2018-09-20:
I like the feedrate icon from the original first post the best. B as second best.

bruce356, 2018-09-20:
I think B is good

dstulken, 2018-09-21:
As much as I like it, I would vote **_against_** "B", as the running person figure matches very closely to the international signage for "emergency exit".
https://en.wikipedia.org/wiki/Exit_sign

On a touchscreen in particular, if common symbology is retained, this could lead to a user misinterpreting this symbol as an emergency stop button, perhaps with disastrous effects in time of need.

My personal favorite would be the original chevron pattern.  :)

alfsoft, 2018-09-21:
I will post more FR variants soon. Really tricky icon to design, as for me. What is feed rate physically? The speed (or ratio?) of filament feeding through the nozzle plus the speed of printer’s kinematics? So what this feed rate (or feedrate?) really is? More like a speed or a ratio? What ‘meaning’ do we need to represent in FR icon?

mcmatrix, 2018-09-21:
B looks fine. 
If you need to make more variants then maybe something like fast-forward icon.

KevinOConnor, 2018-09-21:
My 2 cents - I'd update the fan, bed, and extruder icons.  I would leave the FR as it is in the current code.  (Perhaps changing it to a single 8x16 graphic instead of a 16x16 graphic.)  I'm leery of graphic B - it could be interpreted as a "speed" icon, but it could also be interpreted as a "run away" icon.

As for the HD44780 graphics, I'd leave them as they are in the current code.  (Given the low pixel density on that display, it's hard to add detail to those icons.)

This is just my current opinion - I'm open to alternatives.

-Kevin

KevinOConnor, 2018-09-21:
> What is feed rate physically?

It's a speed override - if it is set to 200% it effectively doubles the velocity of the incoming g-code stream.  Many people never use it.  If there was other pressing information to be displayed, it could be removed from the screen (for example, the fan section could alternate with the FR section every few seconds if FR is not 100%).

FWIW, I'd say that extrude factor (which isn't displayed) has about the same general display priority as feed rate.

-Kevin

marcio-ao, 2018-09-21:
> It's a speed override - if it is set to 200% it effectively doubles the velocity of the incoming g-code stream. 

Perhaps in Klipper it is different, but in Marlin what the FR does is it multiplies the feedrate value of incoming G-Code commands. This, however, is only slightly related to the actual print speed, because the maximum velocity is limited by accelerations and jerk parameters. I did some testing with Marlin that shows that 300% FR roughly doubles the speed of a print, while 900% only quadruples the speed. So as you increase the FR, you get diminishing returns as the limiting factor becomes the accelerations.

Personally I like icon B or D. I understand the concern brought up by @dstulken regarding B, but I think the significant element of the exit icon is a running man set against a door frame, not the man itself. I don't suspect that a running man *by itself* should bring to mind "stop print", especially since it is right next to a 100%.

The FR calls to mind the "turbo" button on old 80s PCs, which I think makes the rocket icon very appropriate, since this is really what it is, a "boost" :)

marcio-ao, 2018-09-21:
> Given the low pixel density on that display, it's hard to add detail to those icons.

My personal favorite has always been the Bill Atkinson icon... the good ol' days where every pixel had to count!

https://www.folklore.org/StoryView.py?project=Macintosh&story=Steve,_Icon.txt

mcmatrix, 2018-09-21:
In `hd44780` we should have one last glyph slot free. 
Maybe we can use it for fan icon?

alfsoft, 2018-09-22:
Some more FR variants. Decided to give this emergency man one more chance and now he is even more speedy! :runner:

[image]

alfsoft, 2018-09-22:
> Perhaps changing it to a single 8x16 graphic instead of a 16x16 graphic

Changing only one icon's width would break visual equality on the screen. All icons are beautifully aligned at this moment because of their equal dimensions and positions.

oderwat, 2018-09-22:
Yep rabbit look like a good choice. For some reason this emergency exit man looks much more like an emergency to me. A racing flag could also do. The rabbit looks maybe a bit to much like a kangaroo. I also didn't recognise the speed-meter in the first images. Maybe having another version of this would work.

bruce356, 2018-09-23:
Hi, the FR replacement should not necessarily just indicate fast speed as in an increase (or going faster) this control is also used to slow down the printer.
Its difficult but the more I look at it the original icon seems to make sense to me, or option "A" speed meter but with a change in dot size, starting perhaps with a medium sized dot at centre, to smaller dots anti clockwise  and lager dots clockwise.
regards bruce

mcmatrix, 2018-09-30:
For `hd44780` fan glyph, what you think of this?
[image]
[image]

KevinOConnor, 2018-09-30:
@alfsoft - FYI - my opinion would be to remain with the current FR icon.  I agree that FR isn't particularly helpful, but I don't think the mental connection of  "picture of something fast" to "speed" to "feedrate" is natural.  I suspect that for all the feedrate icons, people who know what it is will know what it is, and people who don't know what it is wont know what it is.  So, my opinion is that there isn't a gain to the change.  Since many people never change feedrate, maybe just leaving that part of the screen blank when it isn't being used would be an improvement?  I do like the new nozzle, fan, and bed icons.

@mcmatrix - I fear that fan looks too close to a swastika.

-Kevin

mcmatrix, 2018-09-30:
@KevinOConnor I tried to do it in the opposite direction, but yeah it was too close.
There's not much playroom in 5x8 pixels :)
Maybe something like this?
[image]
[image]

KevinOConnor, 2018-09-30:
@mcmatrix - I like it.

-Kevin

KevinOConnor, 2018-10-16:
Let me know if there are further comments on this topic.  Otherwise I'll close it.

-Kevin

alfsoft, 2018-10-17:
@KevinOConnor nothing to add yet. I will open another topic if some new ideas will come :)

closed 2018-10-22 by KevinOConnor