Klipper3d/klipper #656: Homing_ G28 (Internal error on command:"G28")

kein0r, 2018-09-16:
Hello

After successful calibration and one successful test print klipper refuses to work any longer. No changes to printer.cfg were done (besides later reducing homing speed later). 
Error: G28 command fails, sometimes I can run 2-3 G28 commands before it fails sometimes its the first command that fails already. It always fails when starting a real print
Preamble of the print:
```
;FLAVOR:Marlin
;TIME:1007
;Filament used: 0.988942m
;Layer height: 0.15
;Generated with Cura_SteamEngine 3.4.1
M82 ;absolute extrusion mode
; info: M303 E0 S200 C8 ; Pid auto-tune

G28 ; Home to top 3 endstops
M140 S60; Start heating up the base
; Autolevel and adjust first layer
; Adjust this value to fit your own printer!  (positive is thicker)
; This default value is intentionally very high to accommodate the
; variety of print heads used with this printer.  Many of you will
; need tiny values like Z0 or Z0.1.  Use feeler gauges to dial this
; in as accurately as possible.
;G29 Z10
```
Initially my homing speed was set to 50mm/s. I tried reducing it to 5mm/s without luck
[klippy_secondPrintG28-2.log](https://github.com/KevinOConnor/klipper/files/2386846/klippy_secondPrintG28-2.log)
[klippy_secondPrintG28-3.log](https://github.com/KevinOConnor/klipper/files/2386847/klippy_secondPrintG28-3.log)
[klippy_secondPrintG28.log](https://github.com/KevinOConnor/klipper/files/2386848/klippy_secondPrintG28.log)

Any idea why I could print once but now klipper refuses to work any longer?

Thanks in advance for your help

jakep82, 2018-09-16:
I'm not sure if my issue is the same, but I'm also getting same error.  I can home X and Y, but not Z.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2387051/klippy.log)

KevinOConnor, 2018-09-17:
@kein0r - what type of host machine are you using?  It appears the host software is taking longer to generate the homing sequence than expected.  If you change HOMING_STEP_DELAY to 0.000001 at the top of klippy/homing.py and then do a `sudo service klipper restart` does the problem go away?

@jakep82 - same question on host machine, and does changing HOMING_STEP_DELAY fix the issue?

-Kevin

jakep82, 2018-09-17:
That fixed it for me.  I'm currently testing a Pi Zero W.

kein0r, 2018-09-18:
Until now I was running on a raspberry 1b(?) or something. Octoprint + Marlin was working well, Octoprint + Klipper showed the problem described above. I put the card in a raspberry 3 and the problem seems gone for now.

jakep82, 2018-10-01:
I just thought I would mention that I reduced the microstepping on my Z from 32 down to 16, and changed my HOMING_STEP_DELAY back to the stock setting.  This also fixed the issue on my Pi Zero, and I've done multiple test prints without issue.