Klipper3d/klipper #657: Issue: Enhanced Delta Calibration object does not slice

micheleamerica, 2018-09-16:
Hi all
I wanted to try the new Enhanced Delta Calibration but I'm unable to slice the calibrate_size object.
I've tried Cura 15.04.6 and the object does not even show up on the screen, and Slic3r 1.2.9 that crashes when I try to load the object.
Has anyone had the same issue? Is this object stl file available somewhere else for download?
Thanks in advance.
Michele America

micheleamerica, 2018-09-16:
Sorry, my error. The file slices correctly.

closed 2018-09-16 by micheleamerica

KevinOConnor, 2018-09-16:
On Sun, Sep 16, 2018 at 12:58:19PM -0700, micheleamerica wrote:
> Sorry, my error. The file slices correctly.

Could you describe what went wrong?  I've gotten a couple similar
reports and I'd like to be able to update the documentation if there
is a common confusion.

-Kevin

emailcraigs, 2019-05-11:
This is old but I've tried to open the .stl file in multiple programs and nothing will open it. Simplify3d crashes or shows nothing with a size of 9999 which cant be resized. Also, the openscad file will not parse in OpenSCAD. It comes back with a Line 7 syntax error.

emailcraigs, 2019-05-11:
So I figured this out. Since I saw no other download mechanism for the file, I was doing a "save as" in windows and it was giving the file a doctype of HTML which was driving the programs crazy. So, I copied the raw code and created an stl file with it and it opens fine now. I'm just posting this in case someone else as dim as I encounters the same problem.