Klipper3d/klipper #658: Load filament menu option not properly working

knoopx, 2018-09-17:
The "Load filament" option does not work properly:

1) First gcode pushes the filament way too fast (in my humble opinion).
2) Then it "retracts" back instead of slowing down and pushing it further.

I guess the problem is not zeroing extrusion length before the second command.

https://github.com/KevinOConnor/klipper/blob/7b6a44e8f93fd6068facc0c4870d1e350eb82218/klippy/extras/display/menu.cfg#L511

mcmatrix, 2018-09-17:
Thanks for the feedback.
Some menus may not be fully tested.
The fast & slow loading filament lengths are different for everyone :)
I'll try to find time for testing this menu.
I'm also open for any recommendations how to make menu better.

KevinOConnor, 2018-10-25:
Any update on this?  Is this still an issue?

-Kevin

derju, 2018-10-30:
I'm experiencing the same and yes it's still a problem. Perfect would be a way to define a fast re- and detract distance and speed as well as a slow speed and distance.
And maybe while in load a set extrusion length to extrude with a click

mcmatrix, 2018-10-31:
> It'll be hard to make filament menu to fit for everyone.
> Everyone probably has a different need for fast & slow loading filament length/speed.

@KevinOConnor Do you have good ideas how to solve this?
Just to make filament menu with standard values and everyone can overload it with own menu.
Or make an additional generic module with constants and filament standard menu will use these as parameters.
It can be made so that when theres no `const.filament_change` module defined, filament menu is not enabled.
```ini
[const filament_change]
fast_length: 200
fast_speed: 600
slow_length: 50
slow_speed: 120
```
All config attributes will be visible in module `get_status`
Then it will appear in menu parameters as 
```
const.filament_change.fast_length
const.filament_change.fast_speed
const.filament_change.slow_length
const.filament_change.slow_speed
```
Just an idea.

mcmatrix, 2018-10-31:
Current gcodes for filament menu.
```gcode
; Load Filament
G1 E200 F1000
G1 E100 F300
; Unload Filament
G1 E-200 F1000
G1 E-200 F1800
```
So zeroing extrusion length after each commands will solve retract issue?
```gcode
; Load Filament
G1 E100 F600
G92 E0
G1 E20 F180
G92 E0
```

derju, 2018-10-31:
Should do the trick, other option would be to switch in relative mode with G91 and at the end back to absolute with G90

I'll test it later if it's enough to modify the menu cfg

KevinOConnor, 2018-10-31:
I'd recommend using something like:
```
G91
G1 E50 F1000
G1 E50 F1000
G92
```
Using relative mode doesn't reset the extruder position.  Also, the default extrude only length is 50mm, so no single extrude command should be over 50mm.

-Kevin

knoopx, 2018-12-20:
@KevinOConnor somehow switching to relative/absolute mode when loading filament makes the next print fail after probing with an "invalid movement, position out of bounds" error. any idea why this might happen?

fredizzimo, 2018-12-20:
I don't think a macro is enough for this, at least not by the macros that are supported currently.

What's really needed is a python module, with a `load_filament`, and `unload_filament` command, which
1. Automatically does as many extrude moves as needed for the given distance, either in absolute or relative mode. Or alternatively temporarily disables the max distance check
2. Restores the mode that was used before calling the command
3. Restores the extruder position to what it was before the command. I believe that's the missing thing that makes @knoopx see his error.

Number 3. is also important if you want to change filament in the middle of the print, and to be able to continue from there.

This macro should probably have at least four configurable parameters, `fast_speed`, `fast_distance`, `slow_speed`, `slow_distance`. And maybe an `extra_distance`, with an additional command `extrude_more`, so you can keep extruding more, until there's no remains of the old filament.

This module would also make the menu item generic, since you could configure all this in your config file.

totalitarian, 2019-04-29:
Can these values be configured? Obviously some people need loading values depending upon bowden length.

vmorrisPolar, 2019-05-21:
> Can these values be configured? Obviously some people need loading values depending upon bowden length.

I'm with you. I have a direct drive setup thats only about 60mm. Running the load filament would shove filament down so fast itd shred filament for 900mm.

drelich, 2019-12-05:
+1 one for the configure option. Got the BMG clone extruder and the initial speeds grind the filament horribly, with the gears making terrible noise. I'd like to have a way to reduce it to something easier on the wheels. Thanks.

bennodxb, 2020-06-20:
Same thing for me, horrible grinding with direct drive. Had to turn the printer off to stop.