Klipper3d/klipper #659: Illegal Z move after mesh calibration

gspitman, 2018-09-17:
[klippy log.zip](https://github.com/KevinOConnor/klipper/files/2389089/klippy.log.zip)

Hello!

I attempted to add pressure advance last night, and I tried to set my z offset to be a little lower to get a better bed adhesion on the first layer.

Now after mesh bed leveling, as soon as the print is about to start, it says there is an illegal Z move out of bounds, which I would assume is just the mesh compensation?  Anyway, I've attached the logs and please let me know if there is anything else you'd like for troubleshooting.

Thanks!

Arksine, 2018-09-18:
So most likely you have a spot in your bed that is below Z=0 relative to your homing position.  You likely need to set your [stepper_z] position_min to a negative number.

gspitman, 2018-09-18:
Thank you! 

That was it.

closed 2018-09-18 by gspitman