Klipper3d/klipper #665: Serial not working, Ubuntu 18.03, MKS Sbase 1.3

robinduckett, 2018-09-17:
So as the title says, have successfully flashed klipper.bin -> firmware.bin to the MKS Sbase

I have connected the Sbase to my Linux machine temporarily before using a Raspberry Pi

I ran the install-octoprint.sh on my machine to install the local klipper service

I was successfully using the Serial over USB connection with Smoothieware

When attaching the USB cable after flashing, nothing new shows up in /dev/tty*

Attached is the klippy.log, configuration and make .config

I have tried enabling CONFIG_USBSERIAL and disabling it to no avail

Any help would be greatly appreciated

[printer.zip](https://github.com/KevinOConnor/klipper/files/2389699/printer.zip)

khseal, 2018-09-17:
Did you compile the firmware for ubuntu 18?
https://github.com/KevinOConnor/klipper/issues/443

AxMod3DPrint, 2018-09-18:
Have you added the pi user to dialout and TTY groups then rebooted the pi?
Sometimes with a RasPi a logout is not enough.

robinduckett, 2018-09-18:
I compiled in Ubuntu 18.04 - I will try on my raspberry pi, although the idea was to run it off my local machine initially.

This is not a problem with serial ports on my local machine, as it works with Marlin 2.0 flashed on the same board (and obviously smoothieware before that)

robinduckett, 2018-09-18:
Yes, this is my thought too. Marlin 2.0 compiles fine, serial connection works too.

The klipper firmware bin that is output is 17-19kb whereas the marlin firmware is closer to 300 - not sure if that's due to the "thin client" nature of klipper vs marlin or if it's too small because it's not built correctly.

This is the only machine I can test on currently so I'll have to build it on an earlier version of Ubuntu, or just use Marlin 2.0

khseal, 2018-09-18:
I can compile the firmware for you.

FHeilmann, 2018-09-18:
try an older version of the arm toolchain. The one in 18.04 is super-new and I, for one, also had massive issues compiling it.

robinduckett, 2018-09-18:
@khseal many thanks, I will try your compiled version although I would like the chance to compile it myself if possible going forward. I think that running klipper with ubuntu 18.04+ as a host is something I want to set up using such things as Intel NUCs to host. Being able to compile on these hosts as new updates come out would be beneficial to what I'm building.

FHeilmann, 2018-09-18:
5.4.x worked for me on the rpi, I think its just version 6 thats cranky

robinduckett, 2018-09-18:
Can confirm the version @khseal uploaded works on my MKS Sbase. I am running it on a CR-10 and apart from some initial pin inversions, it appears to be running great!

KevinOConnor, 2018-09-18:
Okay - this sounds like a duplicate of #443 then.

FWIW, I regularly compile klipper with newer versions of gcc.  (I'm using gcc v7.1 on my Fedora desktop now.)

As a total guess, I know in the past Ubuntu has done funky things to their compilers with stack checking.  It's possible they enabled that on their embedded compiler (which would be dumb).  If so, the gcc compiler args may variously need one or more of: -nopie -fno-pie -fno-stack-protector -fno-stack-protector-all -fstack-check=no

It will take some time for me to setup an Ubuntu VM to track this down.

-Kevin

mcarpenterjr, 2018-09-25:
Seeing this also after upgrading to 18.04, `/tmp/printer` is not getting created when the device is connected.

KevinOConnor, 2018-09-25:
I'm going to close this if favour of the existing issue #443.

@mcarpenterjr - your issue does not sound like it is the same as this one.  You'd need to create a new issue and attach the klipper log file as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

closed 2018-09-25 by KevinOConnor