Klipper3d/klipper #667: dual x carriage [extruder] not working, [extruder1] is fine

SrGeek, 2018-09-18:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2390645/klippy.log)
I get [extruder] to temperature but the extruder motor does not move.  I flipped extruder motors between E0 and E1 and E0 works fine in the E1 position.  So the motors are fine.  I check continuity of the wires at the Ramps connection and it tests fine. Extruder1 is working fine when I get to temperature.  I hope the attached Klippy.log can shed some light.  Thanks.

KevinOConnor, 2018-09-18:
The log indicates Klipper got a "T0" followed by a "G1 E50".  If you want to extrude on the second extruder you need to send a "T1" followed by an extrusion.  If that doesn't work, issue a "T1", a "G1 E50", an "M112", and then attach the klipper log file from that session.

-Kevin

KevinOConnor, 2018-09-18:
Ah, okay.  Did you run the config checks described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md ?  Remove the filament from the extruder and make sure STEPPER_BUZZ works correctly for it - if not, follow the steps in the link above.

-Kevin

SrGeek, 2018-09-19:
I found the source of the problem: wires were continous but in the wrong order when connected from 4 pins @ramps board to x6 pin @stepper connection.  rearranging wires it will work now.

closed 2018-09-19 by SrGeek