Klipper3d/klipper #669: Printer keeps on pausing during prints

LazaroFilm, 2018-09-18:
My printer keeps on pausing for about 1 second randomly during print. 
I have a Pi3 and Ramps 1.4 on HyperCube Evolution (CoreXY)
According to the graph the Host buffer is saturating?? 
[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/2394110/klippy.log)

jakep82, 2018-09-18:
In my experience this is an Octoprint issue.  It can't process and send the gcode fast enough, particularly when you printing circles with a lot of short line segments.  The solution is enabling the virtual SD feature in Klipper which allows Klipper to read the gcode directly, and removes Octoprint from the middle.

LazaroFilm, 2018-09-22:
Ok, I used the Virtual SD and indeed it fixed the pausing. Now, I lose all the awesome aspect of Octoprint when I do that. Should I report a bug on Octoprint? Is there something else that could be done to help speed the code sent to avoid the pauses? Is there a speed setting in Octoprint to modify? I'm printing at 60mm/s which is not a ludicrous speed.

jakep82, 2018-09-24:
This has been a known issue in Octoprint for well over 4 years.  The only solution right now is to use the virtual SD feature.  Their was discussion about a possible fix coming in 1.4.0, but that conversation is nearly a year old.

https://github.com/foosel/OctoPrint/issues/450

jakep82, 2018-09-24:
I should also note leaving the issue open here won't accomplish anything.  The problem is with Octoprint, not Klipper.  Many people running Marlin also have this problem with Octoprint.

jimfunk, 2018-09-27:
I recently had this issue on a much faster CPU than on the Pi3 (Pentium E5700). It was definitely Octoprint, since it was consuming a full CPU core while klippy consumed about 20%, but I solved it by disabling most of the plugins I had enabled. I suspect it might have been a print time estimator, but I can't be sure since I disabled a bunch of them.

LazaroFilm, 2018-09-28:
Okay. That’s too bead. Maybe we should open a ticket on the Octoprint side. If it’s not already there. I’ll close this for now.

closed 2018-09-28 by LazaroFilm

ddacunha, 2021-06-22:
I confirm that in my case, it is an OctoPrint issue when printing cylinders in vase mode. The issue appeared after installing [DisplayLayerProgress](https://plugins.octoprint.org/plugins/DisplayLayerProgress/) and caused pauses generating droplets effect on the cylinders. Removing the plugin fixed the issue.