Klipper3d/klipper #671: Junction_deviation and square_corner_velocity

swatti87, 2018-09-19:
Sorry for being n00b about this but can "jd" be replaced with the square_corner_velocity?
Or do I need both? 

I can't really see any noticeable difference in corners regardless of my jd-setting but certain numbers lead to failed prints.
Square_corner_velocity on the other hand sounds quite simple in comparison.

AxMod3DPrint, 2018-09-19:
As of commit 0025fbf junction deviation was replaced with  square_corner_velocity so you should only be using square_corner_velocity as JD will not work.

swatti87, 2018-09-21:
I have recently learned my klipper is bit of a "custom job" made by wizards who helped me get my rig klippered.

Looks like it's best to update to the official version.

closed 2018-10-01 by swatti87