Klipper3d/klipper #673: CR-20 Support

urish, 2018-09-19:
Would it be possible to add CR-20 support?

The printer is similar to Ender-3, though as far as I understand, the PCB is not exactly the same...

The printer's stock firmware source code (Marlin based) is available here: https://www.creality3d.cn/?do_action=action.download&DId=56

KevinOConnor, 2018-09-25:
I'd guess that it could be supported.  It doesn't look like anyone has done it already though.  To add support, one would need to determine what electronics the machine has and the board schematics.   From there, one should be able to create a config file - the existing ender3 and cr10 config files may be a good starting point.

-Kevin

urish, 2018-09-25:
Thank you Kevin, do you want me to try working on a PR for the CR-20?

KevinOConnor, 2018-09-27:
If you have the printer and are able to put together the config and test it, then that would be great.  A PR request with the config would be great.

-Kevin

urish, 2018-09-27:
Alright, PR it is!

urish, 2018-09-27:
From my initial testing, it seems like the board is identical to the cr10s. 

I haven't had a chance yet to compile and upload the firmware to my printer, but looking at some photos I found it seems like they use the same board, and I also compared the pin assignments in the c10s config and cr-20 stock firmware and they seem to match. 

I will update here once I get a chance to actually test it on my printer.

KevinOConnor, 2018-10-11:
Thanks.  I'm going to close this for now.  I can reopen if there's further questions.  Feel free to submit a pull request if a change is needed.

-Kevin

closed 2018-10-11 by KevinOConnor

urish, 2018-12-15:
Update: this is now supported as #998 has been merged