Klipper3d/klipper #674: Service Defaults not refreshed during update install - silent failure

adamnock, 2018-09-20:
Quick Note: This is super low priority, its just an issue which might save some users some time if addressed

Issue: When updating the Klipper software the /etc/defaults/klipper file is not correctly updated with the current location

Compounding issue: Service startup fails silently, and due to the log file being a exec argument there is no output in the expected location

Reproduction: 
1) Checkout and install Klipper. Ensure it is running. 
2) Checkout klipper in a different location, install. Remove the old installation location. Startup fails silently.

Like i said, its not a priority by any stretch, but the silent nature of the issue makes is a pain to troubleshoot, so given this should be an easy fix (update the defaults file on install) it should be a quick solve.

KevinOConnor, 2018-09-26:
The /etc/defaults/klipper file is not updated on purpose - it's felt that if the user customized that config file then the update should not modify those customizations.  I can understand where that may be confusing, but I think the alternative could be equally confusing.

-Kevin

KevinOConnor, 2018-10-08:
I'm sure it's possible to improve the install script.  The target audience of the script is users running a RaspberryPi with OctoPi and following the installation instructions - users should not have an issue in that situation.  I'm open to changes, but it isn't something I plan to work on myself.

-Kevin

KevinOConnor, 2018-10-20:
I'm going to close this for now.  I can reopen if someone is interested in doing the development.

-Kevin

closed 2018-10-20 by KevinOConnor