Klipper3d/klipper #682: Make Klippy a proper Python package

Depau, 2018-09-22:
Klippy's code doesn't quite follow Python's packaging and distribution interoperability standards, it's just a directory with a bunch of scripts at the moment.

These are the issues:

- Python 3. I've seen #14 and quite frankly I think that's crazy ([link](https://www.debian.org/doc/packaging-manuals/python-policy/ch-python3.html), [link](http://python3statement.org/), [link](https://www.archlinux.org/news/python-is-now-python-3/)); I don't agree with it but it's your choice. If you like it I can send you a PR that adds support for **both** Python versions with the same source code (as far as I've seen you only need some aliases here and there, not a big deal, not hard to maintain)
- No `setup.py` script
- `/klippy` directory is not a package. You run Klippy with `python2 klippy/klippy.py`. This means the code can't be easily turned into a distro package or installed in a virtualenv. That can be easily fixed by
  - Making all imports under `/klipper` relative
  - Adding a `setup.py` setuptools script configured to automatically add an entry point for Klippy (I already have this one locally)

Once all of this is done, the install scripts should be updated to install the klippy package inside of the virtualenv. This makes it available under `venv/bin/klippy`.

I can take care of everything but I would like to hear your opinion first, before spending time to work on it then seeing my PRs closed.

----

### Meta-issue - Why I filed this issue

I would like to add an Arch Linux package for Klippy, and installation instructions for Arch Linux. I'm not willing to put my face on a package that installs stuff the way the Raspbian installation script does, and would gladly avoid filling the packages with patches that can be easily upstreamed for anybody to profit.

### Personal tip (to painlessly maintain multiple Python versions)

If you have tests, a free CI like Travis will help you catch changes that break compatibility with some Python version immediately with little to no effort.

Depau, 2018-09-23:
At the end I decided to do it anyway.

See the comparison https://github.com/KevinOConnor/klipper/compare/master...Depau:setuptools

All the issues mentioned above have been addressed. I took some extra care to ensure the current way of running it (by running `klippy/klippy.py`) still works on both Python major versions, while adding setuptools and support for ahead-of-time build of the C extension module (a must have for a distro package - you can't just build it at runtime)

I would also like to send a PR for coding style (reformat everything according to PEP styling guidelines), but that may cause a lot of merge issues with other PRs and *it may just be your style*. So let me know if you think that's okay.

KevinOConnor, 2018-09-25:
I'm open to making the code simultaneously compatible with python3 and python2 as long as it doesn't notably impact performance and doesn't result in a maintenance burden.  In such case, I'm open to (and would actively suggest) having a TravisCI python3 test case.

I've tried to keep the code mostly python3 compatible - the big issue is the unicode changes in python3 - I'm not excited by the prospect of littering the code with pointless b'' wrappers, nor am I excited in taking a pointless performance hit by littering encode()/decode() calls throughout the code.  I suspect that will be the biggest issue with having simultaneous python3/python2 support.

It's not clear to me what the real-world advantage of using a python package is.  I'm open to changing the code (with the same caveats as above).

As for the arch package - FYI, Klipper does have some real-time requirements (it must be able to respond to some events within a few hundred milliseconds), so I'm reluctant to recommend using Klipper on a general purpose computer.  I feel combining Klipper with a low-cost single-board-computer is a better total proposition.  That said, I'm all up for improving portability of the code.

It's unclear what you are referring to with the PEP style reference - as far as I know the code basically follows the standard python guidelines (with maybe a handful of slight differences).  If you have some specific changes, best to point them out with some example patches.

It's hard to comment on the full branch diff that you provided.  Each individual change would need to be broken out into a series of separate commits (it looks like you've mostly done this in your local repo).  Note that the recent SAVE_CONFIG changes will likely cause a merge conflict.

-Kevin

oxomoxo, 2018-09-29:
Hi Kevin
Wrt Arch package :
"I feel combining Klipper with a low-cost single-board-computer is a better total proposition"
-> I'm currently trying to setup a Pine64 SBC with an octoprint/kilpper combo running on Arch linux 
It seems I can't find any convenient way to install killpper (octoprint went pretty well) ... I'd be willing to follow an (even not so convenient, like manually installing dependencies etc.) installation guide but I can't find any ...
Is there some place I should start looking to make my way ? Or would there be some serious obstacles that I can't foresee right now ?
Thanks for any info ...
-Lorenzo

KevinOConnor, 2018-09-30:
@oxomoxo - see the FAQ - https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3

-Kevin

KevinOConnor, 2018-10-16:
I'm going to close this for now.  Feel free to submit pull requests as discussed above.

closed 2018-10-16 by KevinOConnor

Depau, 2018-10-16:
I'm sorry i haven't replied yet, I've been snoozing the email for a while and would like to take some time to address everything thoroughly. Hopefully I will get back to it within the next few months as soon as I have enough free time.

Regards