Klipper3d/klipper #684: printer just stoped in mid print

cosminr86, 2018-09-23:
my printer had just topped in mid print
klippy log file attached.
also, i have an about 0.5 sec stop after the first layer
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2408856/klippy.zip)

cosminr86, 2018-09-23:
it seems i have hardware problems with my thermistor.

the problem with the small pause after first layer is still there, if someone wants to take a look

cosminr86, 2018-09-24:
I am wrong again. i was thinking it is a hardware problem, but my part cooling fan was cooling the hotend at 100%. i saw temp drops at 100% fan speed. i lowered at 50-60% and the temp was ok, but i got another print that failed after 10h of print.

i was looking at the log file, and i saw this line

`Stats 475489.8: gcodein=23923542 mcu: mcu_awake=0.116 mcu_task_avg=0.000277 mcu_task_stddev=0.000258 bytes_write=105541002 bytes_read=13350874 bytes_retransmit=2083 bytes_invalid=0 send_seq=1871651 receive_seq=1871648 retransmit_seq=1871651 srtt=0.011 rttvar=0.003 rto=0.100 ready_bytes=2108 stalled_bytes=12974 freq=16000119 heater_bed: target=60 temp=60.0 pwm=0.218  print_time=36998.554 buffer_time=0.002 print_stall=0 extruder: target=235 temp=234.7 pwm=0.716
Heater extruder not heating at expected rate
Lost communication with MCU 'mcu'`

i never had problems before

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2410479/klippy.zip)

KevinOConnor, 2018-09-25:
The root error is "Got error 0 in read: (0)Success" and "Timeout with MCU 'mcu'" - this indicates that the usb connection between host machine and micro-controller was lost.  Things to look for in these cases are bad usb cables, voltage issues, make sure the 5v rail is not shared between host and mcu, etc.

-Kevin

cosminr86, 2018-09-25:
I think I finally found where the problem was. The cables from pi to the relay that power my printer was lose, and had intermittent on/off cycle.

What about the delay after finishing first layer?

KevinOConnor, 2018-09-26:
> What about the delay after finishing first layer?

Generally speaking, Klipper will only delay if the incoming g-code causes it to delay (eg, an explicit delay, or a move/retraction that is slow enough to appear to be a delay).  If you aren't able to track down the source, then issue an M112 immediately after the delay and attach the klipper log file.

-Kevin

cosminr86, 2018-09-28:
it seems this was causing the delay

```
[fan]
kick_start_time: 0.50

```

closed 2018-09-28 by cosminr86