Klipper3d/klipper #685: ./klipper/scripts/install-octopi.sh execule fail in raspberry pi

WindoC, 2018-09-23:
I fail at the step ./klipper/scripts/install-octopi.sh during installation. 

apt-get install virtualenv stm32flash libnewlib-arm-none-eabi gcc-arm-none-eabi binutils-arm-none-eabi return unable to locate package.

[image]

Any idea to fix that?

Thanks!

KevinOConnor, 2018-09-25:
That's certainly odd.  You should make sure the OctoPi image is pristine (no download errors) and retry the OctoPi install from scratch.  Make sure your internet connection is working well when running the install-octopi.sh script.

-Kevin

WindoC, 2018-09-25:
So it must use the one download from the link https://octopi.octoprint.org/latest . Right?

Are there any guide to install it from the default raspberry pi image. I Would like to use klipper but not the OctoPi.

Thanks!

KevinOConnor, 2018-10-01:
There's some discussion on this at: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3 .  Lots of people run Klipper on something other than OctoPi, but to do that you'll need to track down all the Linux admin steps necessary to make it work.

-Kevin

closed 2018-10-01 by KevinOConnor