Klipper3d/klipper #686: `M73 P101` aborts the print and disconnects

cassianoleal, 2018-09-23:
I'm at the end of a print and I imagine an Octoprint plugin injected a `M73 P101` command.

Since `M73` is to set print progress, Klipper recognises `101` as an invalid value and aborts the print.

I understand taking the defensive approach and stopping a print when certain parameters are out of bounds but `M73` is merely informative and should not cause an abortion.

```
Send: N611 G1 X174.569 Y103.198 E1.11748*100
Send: N612 G1 X174.569 Y102.663 E0.02037*105
Recv: ok
Send: N613 M105*35
Recv: ok
Recv: ok T0:234.2 /235.0 B:75.0 /75.0
Send: N614 M73 P101*84
Send: N615 G1 X154.337 Y82.431 E1.08868*84
Recv: !! Error on 'N614 M73 P101*84': P must have maximum of 100
Changing monitoring state from "Printing" to "Error: Error on 'N614 M73 P101*84': P must have maximum of 100"
Changing monitoring state from "Error: Error on 'N614 M73 P101*84': P must have maximum of 100" to "Offline (Error: Error on 'N614 M73 P101*84': P must have maximum of 100)"
Connection closed, closing down monitor
```

KevinOConnor, 2018-09-26:
Should be fixed now (commit 170b9678).

-Kevin

closed 2018-10-01 by KevinOConnor

BoHoehne, 2020-01-25:
I just had the M73 abort problem again.

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4112531/klippy.log.zip)
[printer_cfg.zip](https://github.com/KevinOConnor/klipper/files/4112532/printer_cfg.zip)