Klipper3d/klipper #689: TMC 2208 shuts down after 1 min of printing

Thesecky, 2018-09-24:
Hi. I just switched to klipper from marlin. Bassicaly everything is the same except from the software. But now, after i try to print a benchy, after the mesh bed leveling is done and the first layer starts to print, first the x axis stops working and after a few seconds y. But that didnt happen on a previous print when i tried to print a simple cube. What is wrong. I have my drivers in standalone mode so stealthchop 2, drivers are at 1v, VM is 24 volts.

jakep82, 2018-09-24:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print

Thesecky, 2018-09-24:
I dont think log would help or would it?
What should i set the square_corner_velocity to. If the dafault is 5, will 4 be sufficient?

jakep82, 2018-09-24:
It's impossible to know if the log file will help until we see the log file.  You can transfer the log file with PSFTP.

Others reporting this same issue have indicated you need a square_corner_velocity of zero.

https://github.com/KevinOConnor/klipper/issues/196

Thesecky, 2018-09-24:
Setting it to 0 also didnt help. I have putty but i only know how to edit the config file nothing else. Really dont know what to do now.

jakep82, 2018-09-24:
Perhaps the stepper drivers are overheating.  Add cooling or turn down the Vref on the steppers.

Thesecky, 2018-09-24:
I have nema 17, 1.68A motors. Now i have 1v, will try 0.9.

Thesecky, 2018-09-24:
Putting the vref to 0.7V and adding another fan also didnt help

jakep82, 2018-09-24:
I can't offer any more help without a log file.

Thesecky, 2018-09-24:
Ok will learn how to do it now

Thesecky, 2018-09-25:
I have, but i dont know how to get to that folder let alone copy it.

Thesecky, 2018-09-25:
Oh this program seems to be alot easier than using putty. I think i will manage now! Thank you

Thesecky, 2018-09-25:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2415536/klippy.zip)
Finaly here is my log. I pressed M112 obout a second afte the y axix failed.

jakep82, 2018-09-25:
This log indicates you have square_corner_velocity set to 5.  Try setting it to 0, issue a restart, and run the same file.

KevinOConnor, 2018-09-25:
This looks like a duplicate of issue #196.

-Kevin

Thesecky, 2018-09-25:
I already did that with no inprovments, only the movement of toolhead became really unsmooth in the first layer of file, before the axix shut of.

Thesecky, 2018-09-25:
So without switching to uart and setting them in spreadclycle, there is nothing i can do. Then i might aswel have bought the 2130 ones if i cannot use the stealthcop2.

KevinOConnor, 2018-10-01:
If you have the 2208s, I'd definitely recommend connecting the PDN_UART pin to a pin on the micro-controller.

In any case, lets move the conversation to the existing issue #196.

-Kevin

closed 2018-10-01 by KevinOConnor