Klipper3d/klipper #690: Timer too close & retransmits on printrboard.

jimthouston, 2018-09-24:
[image]

This is a follow up for my previous issue #594.

In my case the only communications errors were the re-transmits.  I added a bunch of debug and saw the re-transmits being sent even though the messages were being acknowledged slowly.  I changed the MIN_RTO value in serialqueue.c to 0.1 seconds and it eliminated the re-transmits.  I'm including a log with my debug and a patch with my changes in case you want to see the debug I added.

With the retransmit problem solved I was able to print a quarter of a 2 cm. cube before getting a "Timer too close" message.  I notice that the last sent message in the dump was a schedule_soft_pwm.  This command sets its timer to the  time when the sensor was read + REPORT_TIME value in thermistor.py.  By changing this delay from 0.3 seconds to 0.5 seconds I was able to print a 2 cm. cube.

For the printrboard where the communications are never really sent over a serial link, the re-transmits in software would only be needed if there is a bug in the serial over USB emulation.  If there are real communications issues the USB hardware would also be doing error checking and re-transmission.

Changing the REPORT_TIME might be OK but might be worth adding an argument to the timer setting code to identify timers which can be allowed to issue late.  The timers which are controlling stepper motors you want to happen on time but the sched_soft_pwm might not need to shutdown the printer if it late.

I would be glad to work on fixes for both re-transmit and timer issues.  What would you like?

[serialqueue-debug-patch.txt](https://github.com/KevinOConnor/klipper/files/2411196/serialqueue-debug-patch.txt)

[klippy-mydebug3.log](https://github.com/KevinOConnor/klipper/files/2411083/klippy-mydebug3.log)

KevinOConnor, 2018-09-26:
It is true that a usb connection should do hardware retransmits, but the symptoms of a bad cable are basically the same - a series of hardware errors will result in a delay that causes the software level to do retransmits.

I don't think changing REPORT_TIME or MIN_RTO is the correct approach.  There shouldn't be a need to tune these values (these values are huge relative to the round trip time of the communications - the only time these values would come into play is if something is drastically wrong with the communications).

That said, it is possible there is a software issue in the AVR implementation of usb serial.  The code is taken from a popular library (see lib/pjrc_usb_serial).  However, there might be an error in the Klipper integration.  I have an at90usb chip I can test on, but it's not something I'll be able to get to in the next few days.

-Kevin

jimthouston, 2018-09-26:
I should say that I'm excited to finally print the 2 cm cube without errors.  Did you see that cube and it is really 2 cm.  I guess the over hang bits could be better...

My guess was that klippy on the host side gets preempted between when it samples and calculates the time and when it sends the schedule_soft_pwm message.  

I don't really understand where the why the retransmit timer is triggering.  Individual message seem to have reasonable round trip times.  I'm attaching a log from an earlier failure before I changed the MIN_RTO value so you can see the timing of the reads/writes and retransmits.  I think the retransmits are happening when the link has gone idle and the host dumps the full transmit window on the MCU.  In my debug I include the event_time values and the offset (monotonic-event_time).  It looks like the ACKs are being buffered and several are delivered together.  I will try try to figure out where this is happening.
I assume that this implies a delay either in the MCU sending or the host receiving.  My guess is the host side.  I experimented a bit with using libusb instead of the linux cdc-acm driver.  This seemed like a good idea when I thought that messages were really being lost.  I noticed that there were EAGAIN messages in the log from the original failure.

I'm confident that it isn't an electrical noise problem.  I eliminated all of the source of noise that I could think of even running the controller board with out a printer.  No stepper motors no heaters switching just a trim-pot to replace the thermistor.  The problem still reproduced.

I'm falling asleep writing this and I hope it makes some sense.

[klippy-mydebug2.log](https://github.com/KevinOConnor/klipper/files/2418074/klippy-mydebug2.log)

KevinOConnor, 2018-09-27:
It looks like I can reproduce this problem on my at90usb1286 test chip.  I suspect there is a problem with the way the Klipper micro-controller code integrates with the pjrc usb library (as a guess, the library may be delaying writes in such a way that it temporarily stalls the data transfer).

I'll take a look at it.

-Kevin

KevinOConnor, 2018-10-01:
I believe this should be fixed now (primarily as of commit d05aa819).  The avr usb code now uses the same usb code that the LPC176x and SAMD21 boards use.  I didn't track down the root cause of the integration with the pjrc library, but I have been able to confirm that I can no longer reproduce the original problem with the common usb library code.

It may be necessary to update the serial port after upgrading to the latest code (see config/generic-printrboard.cfg for the new usb serial port).

@wizhippo - can you verify everything still works on the at90usb646?

-Kevin

jimthouston, 2018-10-01:
Thank you.  This fixes the problems I was seeing.  It printed a 2cm cube without retransmits.

closed 2018-10-08 by KevinOConnor