Klipper3d/klipper #695: Bed heatup is stopped

Galunid, 2018-09-24:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2412738/klippy.log)
Hello, after installing software, I wanted to jump straight into printing. Unfortunately I have only ABS at hand. I decided to give it a try. When I tried to heat the bed to 90 *C, as I usually do, I noticed a problem. After a few minutes bed stopps heating(heat up temperature drops to 0 in octoprint). After some digging I've found `MAX_HEAT_TIME = 5.0` variable, which I believe is responsible for this. The question is whether it is safe to change it, as it's used to set these up
`./heater.py:        self.mcu_pwm.setup_max_duration(MAX_HEAT_TIME)
./heater.py:        self.next_pwm_time = pwm_time + 0.75 * MAX_HEAT_TIME
`
Another thing I noticed, is that the bed heats noticably slower, than when I used Marlin(I moved pid settings from Marlin). Is there some kind of 'lock', that prevents bed from drawing more current? I have 12V/33A power supply, so I believe it'd be safe, to increase it. I tried to add `max_power: 1.0` to `[heater_bed]` config tab, but it didn't do much of a difference. 
Thanks for developing this cool software, Galunid

Flupke1979, 2018-09-24:
Hello:

Check the config_checks documentation:
PID settings found in other firmwares or in the example configuration files often
work poorly.

You need to do the PID calibration.

Kim

Galunid, 2018-09-24:
Well, thanks for your reply. Will I be able to do pid calibration, when I'm not able to reach desired temperature at all?

Flupke1979, 2018-09-24:
Did you check if the thermistor settings are correct? If those are correct you could just give it a try.

Galunid, 2018-09-24:
Yes, I confirmed that the thermistor is correct

jakep82, 2018-09-24:
Try running the bed in watermark mode instead of PID.

Galunid, 2018-09-24:
@jakep82 It's even worse than before. The primary reason why I changed from watermark to pid, was because the bed couldn't heat up

jakep82, 2018-09-24:
Watermark is either on or off, nothing in between.  If it can't reach temperature with watermark, then you have a hardware issue.

jakep82, 2018-09-24:
Not sure what to tell you.  Your log indicates you were sending full power to the bed and it wasn't able to heat it above 81C.  This doesn't seem like a software bug.

Galunid, 2018-09-24:
As I mentioned earlier, heating was interrupted for some reason after ~5 minutes.

KevinOConnor, 2018-10-01:
The log indicates the heater got turned off after 10 minutes.  This is likely due to the idle timeout - see config/example-extras.cfg for the details.  You can increase the idle timeout, redefine the commands run on an idle timeout so that it does not run TURN_OFF_HEATERS, or use a command like M190 to wait for a particular temperature so that the code knows the printer is not idle.

-Kevin

closed 2018-10-08 by KevinOConnor