Klipper3d/klipper #696: Can't make the bed level

Avalonnw, 2018-09-25:
Hi,
I'm looking for some help with my delta.
I tried both main branch and enhanced calibration and the problem is the same: my bed is not flat and level.
Even when I do mesh levelling and it's all nice and good, manually jogging down results in right and back sides about -0.05 - -0.15 below the nozzle (on a paper test).
What am I missing? I was under impression that mesh bed levelling should take care of any residual irregularities and issues with calibration AND uneven bed. What is strange also, is that I use Ultrabase, which is glass, and it's supposed to be even and flat by itself.

Arksine, 2018-09-25:
Mesh leveling should accommodate for bed irregularities, however there are some things to keep in mind:

- Mesh Leveling doesn't currently have great support for circular beds, as currently it is only capable of probing and interpolating a square mesh.  Work is being done toward better support for circular beds.

- Currently Mesh Leveling does not persist between restarts.  This feature will be added soon, however currently BED_MESH_CALIBRATE needs to be run each time the printer restarts for mesh generation and adjustment.

Avalonnw, 2018-09-25:
I understand the limitations, however even running mesh lvelling a few times gives vastly different result.
I just tried a test print and basically ruined my ultrabase:
[image]
[image]

Galunid, 2018-09-25:
A bit offtopic, but I think you can still clean it. It just looks like plastic was pushed into the pores of the ultrabase. Try using something to dissolve the plastic(for example aceton if printed with abs). There's a good change that it's not destroyed

Arksine, 2018-09-25:
If the probed z values are wildly inconsistent when doing BED_MESH_CALIBRATE then it sounds like the printer itself is out of calibration.  I assume you have run DELTA_CALIBRATE and updated the endstops/angles accordingly?

I hope @Galunid is correct and you are able to restore your bed surface.  I would recommend attempting to get consistent results without using mesh leveling before making another attempt with it.   If you continue to have issues please include klippy.log and octoprint.log.  If making another attempt with Mesh Leveling please run the BED_MESH_OUPTUT gcode after running BED_MESH_CALIBRATE so I can see what the probed points and generated mesh looks like.

Avalonnw, 2018-09-25:
@Arksine I did run both DELTA_CALIBRATE and enhanced calibration. I'll try to square things up a bit more but I was sure between DELTA_CALIBRATE and BED_MESH_CALIBRATE any inconsistencies/irregularities in structure would be taken care of. I heard someone printed on an Easter egg even!

@Galunid Thanks for the suggestion, I'll try but I don't think it will help - I can actually feel indents with my finger. Luckily, I've got another UB, which I will use once I figure out the BED_MESH_CALIBRATE thingy.

KevinOConnor, 2018-09-25:
> I'm looking for some help with my delta.
I tried both main branch and enhanced calibration and the problem is the same: my bed is not flat and level.
Even when I do mesh levelling and it's all nice and good, manually jogging down results in right and back sides about -0.05 - -0.15 below the nozzle (on a paper test).
What am I missing? I was under impression that mesh bed levelling should take care of any residual irregularities and issues with calibration AND uneven bed. What is strange also, is that I use Ultrabase, which is glass, and it's supposed to be even and flat by itself.

If you're working with a truly flat surface then I don't think you should use BED_MESH_CALIBRATE - this is something that DELTA_CALIBRATE should handle.  I recommend you update to the latest code, rerun DELTA_CALIBRATE using the manual (paper test) method, run SAVE_CONFIG, and identify any position that has significant deviation (provide the reported X,Y,Z position and the actual Z position).  Then attach the full Klipper log file as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Avalonnw, 2018-09-25:
@KevinOConnor I will try that tonight, thank you.

ruevs, 2018-09-26:
Mesh bed leveling should not make things worse in theory. But in practice if the flatness of the bed (e.g. glass) is better than the precision/repeatability of the leveling sensor it can make things worse.

In addition compensating for imprecise geometry or loose mechanics by "calibration" is not optimal. Try to check that there are no loose screws, ball joints, belts. Check how much you can "rock" the head by hand when the motors are on (e.g. after homing) if it wobbles more than a fraction of a millimeter you will not get consistent results.

The six arms should be as close as possible to each other in length - this is very important and hard to compensate with calibration.

The three towers should be as close as possible to parallel to each other (and perpendicular to the bed) - this is also hard to compensate for by calibration. In tother words the top and bottom triangles should be equilateral, parallel to each other and exactly aligned.

Avalonnw, 2018-09-28:
@ruevs It's all fine, I understand that ideally everything should be nice and square and equal etc. No point in calibration then.
"A precisely built delta printer with a flat bed does not need mesh bed leveling" nor calibration. However there is no such thing as a precisely built delta unfortunately. Hence, Calibration and Bed levelling.
Interestingly, same printer with Marlin calibrates differently and after bed levelling works ok. There are other issues with Marlin though, and I don't want to use it really.

Avalonnw, 2018-09-29:
Ok, so without bed levelling it works, sorta. First, the bed is not flat, so there are high and low points, and bed levelling does not resolve that. Second, I've got a separate issue with TMC2130 skipping steps randomly... Arrrgh! Any advice on TMC programming?

Hywelmartin, 2018-09-29:
@Avalonnw 24v instead of 12v(that is out of spec for the tmc drivers)

Avalonnw, 2018-09-29:
Really? Datasheet says differently: https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2130_datasheet.pdf

Hywelmartin, 2018-09-30:
@Avalonnw sorry for that(out of spec, damned youtubers =)) But they work so much better with higher voltage

KevinOConnor, 2018-10-08:
I'm going to close this as I think the original question has been answered - bed mesh needs some additional support for round beds before it's useful on delta printers so it is best to stick with just regular delta calibration for now.

-Kevin

closed 2018-10-08 by KevinOConnor

Avalonnw, 2018-10-09:
@KevinOConnor 
I have updated my klipper, ran delta calibration with a piece of paper and the result was pretty much the same. I've also re-made my sensor (just in case it was faulty after driving it into bed so many times) and re-calibrated it all again. Result is the same. With AND without bed levelling the picture is similar (see attached photo). It looks like bed levelling does not do anything in my case which makes me think something is wrong with my settings.
Another thing, even with fade_start set to 1mm, whenever I issue G1Z0 I get Z0.03 or similar.
[image]

KevinOConnor, 2018-10-10:
Looks like poor delta parameters to me.  I recommend you clear out any SAVE_CONFIG parameters from the config file, don't enable bed_tilt or bed_mesh in the config file, run DELTA_CALIBRATE using manual mode, perform the "paper test" at the 7 points, then run SAVE_CONFIG, and finally attach the full klipper log file from the above session.  If after doing that you have adhesion issues you should post some pictures of prints you attempted.

-Kevin

Arksine, 2018-10-10:
@Avalonnw fade_start is the height at which z-adjustment starts phasing out, so at Z0 bed_mesh is looking up the z-value of the mesh at the corresponding x,y coordinates and applying it.  If the toolhead is at the origin it is possible for a small adjustment to be applied due to probing inaccuracies, interpolation, etc.

I can't be sure why bed_mesh isn't making adjustments without logs and the results from BED_MESH_OUTPUT, but as Kevin suggested its probably best to do without mesh leveling on deltas right now given how small of an area the mesh can represent.

amalabey, 2018-12-28:
I have the same issue with Anycubic Kossel (Upgraded Pulley) with Ultrabase. I followed the suggestions in this thread and did only a basic manual calibration, but the print head crashes on to the bed when it moves to one side of the bed. See attached [printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2715825/printer.cfg.txt) and [klippy.log](https://github.com/KevinOConnor/klipper/files/2715824/klippy.log). Can you please re-open this issue.

[image]

amalabey, 2018-12-30:
Ignore the issue above, it was caused by friction in one of the belts.