Klipper3d/klipper #701: Incorrect Z calculations with BED_MESH_CALIBRATE and G92

alecplumb, 2018-09-27:
After doing a BED_MESH_CALIBRATE, then a G28, then a G92Z0 (after finding the correct 0), the Z calculations are incorrect.

```
Send: G92Z0
Recv: ok
[...]
Send: get_position
Recv: // mcu: stepper_x:2838 stepper_y:6346 stepper_z:375 stepper_z1:467 stepper_z2:454 stepper_z3:404
Recv: // stepper: stepper_x:340.000000 stepper_y:-0.000000 stepper_z:-1.622500 stepper_z1:-1.622500 stepper_z2:-1.622500 stepper_z3:-1.622500
Recv: // kinematic: X:170.000000 Y:170.000000 Z:-1.622500
Recv: // toolhead: X:170.000000 Y:170.000000 Z:-1.622604 E:-800.000000
Recv: // gcode: X:170.000000 Y:170.000000 Z:1.800000 E:-800.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:1.800000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok

// carriage ~6mm above stop

Send: G1Z10
Recv: ok
[...]
Send: get_position
Recv: // mcu: stepper_x:2838 stepper_y:6346 stepper_z:5744 stepper_z1:5836 stepper_z2:5823 stepper_z3:5773
Recv: // stepper: stepper_x:340.000000 stepper_y:-0.000000 stepper_z:11.800000 stepper_z1:11.800000 stepper_z2:11.800000 stepper_z3:11.800000
Recv: // kinematic: X:170.000000 Y:170.000000 Z:11.800000
Recv: // toolhead: X:170.000000 Y:170.000000 Z:11.800000 E:-800.000000
Recv: // gcode: X:170.000000 Y:170.000000 Z:11.800000 E:-800.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:1.800000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000

// carriage ~19mm above stop
```
`stepper_z*` positions seem to accurately reflect the change in height (about 13mm), while the `gcode`, `toolhead`, and `kinematic` positions think it's only 10mm.

The result is that printed objects that should be 10mm high are actually 13mm high. The deviation maxes out at 10mm (e.g., a 25mm object is 28mm)
[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/2424859/klippy.log)

aeropic, 2018-09-27:
I don't know if this is linked to your problem but I had too an incorrect Z height due to the fact I add not set the two following parameter with the same value:
[stepper_z]
endstop_pin: probe:z_virtual_endstop
position_endstop: 2.45  <<<<<<<<<<<<<<<<<<<<

[probe]
z_offset: 2.45 <<<<<<<<<<<<<<<<<<<<<<

For instance if you put 0 to position endstop while z_offset is left at 2.45, you will get a normally 20mm cube printed with a 22.45 mm height ...

alecplumb, 2018-09-27:
It's probably worth noting that this is a Voron 2, where the Z endstop mechanical switch is separate from (has has a different offset from) the inductive probe used for bed probing.

Arksine, 2018-09-28:
@alecplumb It sounds to me like the probed coordinates must be quite a bit higher than expected and the Mesh is referenced to around Z=3.  What is happening is that BED_MESH is applying an adjustment, phasing out until gcode Z=10.  The result is that 3mm is added to the print, because BED_MESH thinks that the whole bed is 3mm too high.  Perhaps you could supply the results of BED_MESH_OUTPUT after probing the bed?

BED_MESH makes a couple of assumptions with regard to the configuration in order to generate a mesh that accurately represents the bed:

1) The supplied z-offset in [probe] is the distance between the bed and the nozzle when the probe triggers
2) The bed is located at Kinematic position Z=0

It seems like in your situation assumption 1 may be satisfied, but not assumption 2.  In that case you will need to either adjust your z position_endstop so that the bed is located at  Z=0 relative to the endstop, or you need to adjust your z_offset to account for the discrepancy.  Essentially, what is required is that the z_offset is able to adjust the probed values so that the mesh is referenced to 0.

alecplumb, 2018-09-28:
You are correct that the problem is caused by the `[probe]` z_offset being incorrect. I had assumed that, like Marlin, `BED_MESH_CALIBRATE` would read the probe values for relative variations rather than as absolute distances to the bed. I have since set the z_offset to "close" to the actual probe trigger distance, but even then the deviation in object height is too high. 

There should be some way to correct the height after doing a calibration that doesn't result in stretching or shrinking Z height.

Arksine, 2018-09-28:
You can disable z fade by setting the fade_end option in bed_mesh to 0.

closed 2018-10-16 by KevinOConnor