Klipper3d/klipper #702: Selecting UART3 on Trigorilla board

mbgroot, 2018-09-27:
i'm use UART3 on Trigorilla to connect Raspberry Pi. 
i change CONFIG_SERIAL_PORT to 3 but after make commant it change to 0 back.
or am I changing the wrong place?

KevinOConnor, 2018-09-28:
Code changes would be required to support UART3 on the atmega2560 (specifically, the src/avr/Kconfig file would need to be changed).

-Kevin

mbgroot, 2018-09-28:
Thanks for the help, but it did not help.
Changed to 3 and put in the printer.cfg _/dev/serial0_ - the host does not connect to the printer.
Ports and cable are working. On firmware MK4duo everything works.

KevinOConnor, 2018-10-02:
I added support for selecting alternate UARTs on the 2560 (commit d348f0da).  You might want to try downloading the latest code, make sure you are using an unmodified copy of Klipper, rerun "make menuconfig", recompile the firmware, and reflash the firmware.

If it still doesn't work after doing the above, then attach the full klipper log file as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

mbgroot, 2018-10-02:
I rerun "make menuconfig", recompile the firmware, and reflash the firmware. But still don't work...

[klippy.log](https://github.com/KevinOConnor/klipper/files/2436696/klippy.log)

KevinOConnor, 2018-10-08:
Alas, I have no further advice.  As far as we know, the code works as intended.

-Kevin

closed 2018-10-16 by KevinOConnor