Klipper3d/klipper #704: Printer just stalls, second time now

dotps1, 2018-09-27:
So my printer has just been stalling.  this time it did it on a huge print, and I basically just wasted a quarter roll of filament.  I have done other long prints before no problem, so I don't know what the issue is.  Octoprint is still responsive, I have tried replacing the U

SB cable.  Log attached.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2425918/klippy.zip)

FHeilmann, 2018-09-28:
```
stepcompress o=4 i=0 c=3 a=0: Invalid sequence
Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 169, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 69, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/gcode.py", line 467, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 350, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 190, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 176, in flush
    move.move()
  File "/home/pi/klipper/klippy/toolhead.py", line 103, in move
    self.toolhead.extruder.move(next_move_time, self)
  File "/home/pi/klipper/klippy/kinematics/extruder.py", line 197, in move
    self.stepper.step_itersolve(self.cmove)
  File "/home/pi/klipper/klippy/mcu.py", line 132, in step_itersolve
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress
```
@KevinOConnor thats the same error I was seing when I tried your suggestion for resume functionality, maybe something is amiss here?

dotps1, 2018-09-28:
it took me a lot of tries to get this working with my ender 3, mainly had flashing issues with the missing bootloader.  Anyway, I thought maybe I messed something up while configuring it, so I backed up my config and rebuilt everything.   Both the OctoPrint server and reinstalled Klipper.  Maybe over kill but I thought I would see if it fixes it.  Like I said it seems random, so I really don't want to waste that kind of time and filament again.

dotps1, 2018-09-28:
I had to re generate the gcode, this would have been extremely close to what I used.  I apologize I don't have the original.  when it stopped the z axis was around 78.6.  (the file is too large and I can't zip it small enough, you can download it from my OneDrive:  https://1drv.ms/u/s!Ai_fryXP5gjlgY4vQIfLuTyx2efdgQ).  Thanks.

KevinOConnor, 2018-09-28:
Okay, that g-code doesn't match, but it does generate a similar error at a different location in the print.  (See https://github.com/KevinOConnor/klipper/blob/master/docs/Debugging.md#translating-gcode-files-to-micro-controller-commands if curious on how to check g-code files.)  I'll take a look, but it will likely take several days for me to track this down.  Don't try to print this file - it will fail every time.

-Kevin

dotps1, 2018-09-28:
sounds good, thanks a lot!  yeah, I printed it a touch smaller and it failed way earlier in the print.  but I don't have that gcode either, I just kept over writing the same name.  thanks again.

dotps1, 2018-09-29:
well, happened again just now, I got to be honest, this is probably the end of klipper for me.  I can't keep wasting this much filament and time to have prints stop randomly.  everything was going fine then it just shuts off:

from the console:
```
Send: N49449 G1 X136.856 Y101.96 E2045.61946*105
Recv: ok
Send: N49450 G1 X141.905 Y101.96 E2045.7538*83
Recv: // Klipper state: Shutdown
Recv: ok
Send: N49451 G1 X141.905 Y112.139 E2046.02465*91
Recv: // Lost communication with MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! Lost communication with MCU 'mcu'
Changing monitoring state from "Printing" to "Cancelling"
Send: N49452 M84*17
Recv: ok
```

here is the log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2430252/klippy.log)

here is the gcode of what I was printing:
https://1drv.ms/u/s!Ai_fryXP5gjlgY4wM2lPUUsUZHuvMQ

thanks.

KevinOConnor, 2018-09-30:
> well, happened again just now

This second error is not related to the first error.  The first error appears to be an issue with the internal Klipper kinematic code, while this is a communication issue.

The root error here is "Got error 0 in read: (0)Success" and "Timeout with MCU 'mcu'" -- this indicates that the host operating system disconnected the usb device attached to the micro-controller.

Things to look at: check your usb cables, make sure there aren't low voltage issues, make sure the host 5V USB power isn't mixed with the micro-controller's 5V power.

-Kevin

KevinOConnor, 2018-10-02:
I've tracked down the original problem.  There was a bug in the handling of the M204 command - if an M204 was used to reduce the acceleration below the configured "max_accel_to_decel" setting then the lookahead code would not function correctly.  This should now be fixed (commit 0820394e).

-Kevin

dotps1, 2018-10-02:
thanks a lot @KevinOConnor!  I will try the print again when get some time.  I will close the issue for now (and re open if come across it again).  As far as the other lockup, I swapped the power cable and redid that print and did not have the issue.  so must have been a bad cable I was using to power the pi.  Thanks.

closed 2018-10-02 by dotps1