Klipper3d/klipper #705: MCU Shutdown

Maximuscr31, 2018-09-27:
Basically this is stopping randomly. Is it a connection issue? Slicer issue? or a Temp issue? I am really not understanding which problem it is in the log file. Thank you for the help on the direction to go. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/2425980/klippy.log)

dotps1, 2018-09-27:
this looks similar to the issue I just posted as well #704

FHeilmann, 2018-09-28:
Just before the MCU times out your heater temperature drops down to 214.1 degrees. The log also indicates that your heater was not running at the expected rate:

`Heater extruder not heating at expected rate`

try printing something without your cooling fan on and see if the issue persists. If that solves it you might have to tweak your heater checks (see https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L274 for an example how to do this)

KevinOConnor, 2018-09-28:
The root error here is "Got error 0 in read: (0)Success" and "Timeout with MCU 'mcu'" -- this indicates that the host operating system disconnected the usb device attached to the micro-controller.  (The minor temperature difference just prior to the disconnect I'd guess is just minor variations in temperature.)

Things to look at: check your usb cables, make sure there aren't low voltage issues, make sure the host 5V USB power isn't mixed with the micro-controller's 5V power.

-Kevin

Maximuscr31, 2018-09-29:
The board I had was a mks sbase clone from amazon with the integrated a4982 drivers. I went ahead and sent it back yesterday since I found the voltage on the extruder was being erratic and bouncing around. I went ahead and just got a cheap mks l and ordered the 2208 drivers. I will go through everything and try to make sure everything is good on the board. I will close this and hope it resolves itself on the next board. Thanks again Kevin for all the help. It really means alot to us that you do this.

closed 2018-09-29 by Maximuscr31