Klipper3d/klipper #708: Extrude only move too long

smark-, 2018-09-29:
Whenever I complete a print (or cancel it) and I try to start a new one I always get the error: " Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Extrude only move too long (-6565.890mm vs 50.000mm). It seems like klipper doesn't restart the extrusion counter after a printer is finished. The only way to keep printing it to RESTART the host.

Flupke1979, 2018-09-29:
Reset your extruder in the G-code startscript of the slicer.

jakep82, 2018-09-29:
G92 E0

smark-, 2018-09-30:
Thanks, this is a good work-around. I was wondering why this feature is not implemented in Klipper already.

KevinOConnor, 2018-09-30:
This is a g-code problem.  There's nothing Klipper can do about it - all firmwares have the same requirement.

-Kevin

LazaroFilm, 2018-10-01:
Even on Marlin you should start your GCode with resetting your extruder, that should be the first line of your start code.

smark-, 2018-10-01:
I might have missed it, since I don't recall having to do so. If that's the case, that's not a klipper issue

closed 2018-10-01 by smark-