Klipper3d/klipper #709: gcode_macro parameters not working

lufton, 2018-09-29:
I have gcode_macro section in my printer.cfg that uses parameter DELAY (from **example-extras.cfg**):
```
[gcode_macro MY_DELAY]
gcode:
	G4 P{DELAY}
```
Now I execute:
```
MY_DELAY DELAY=100
```
And get:
```
Recv: !! Error on '"G4 P{DELAY}"': unable to parse {
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/2430919/klippy.log)

mcmatrix, 2018-09-30:
From your log I see that your macro definition is
```ini
[gcode_macro MY_DELAY]
gcode = "G4 P{DELAY}"
```
Try to remove double quotes
```ini
[gcode_macro MY_DELAY]
gcode = G4 P{DELAY}
```

lufton, 2018-09-30:
I think I tried both and both is not working.

lufton, 2018-09-30:
Actually I tried:
```
gcode:
G4 P{DELAY}
```

mcmatrix, 2018-09-30:
It seems that you don't have the latest code.
Try to update your codebase.
```
cd ~/klipper
git pull
sudo service klipper restart
```

I get the same parsing error when I use `gcode_macro.py` before the `macro improvement with parameters` change.

mcmatrix, 2018-09-30:
Check the size of `gcode_macro.py`. It has to be ~1307 bytes.

lufton, 2018-09-30:
Yeap, ```git reset --hard``` helped. Sorry. Closed.

closed 2018-09-30 by lufton