Klipper3d/klipper #710: FORCE_MOVE problem

lar3ry, 2018-09-30:
Tried to do a FORCED_MOVE with the command
FORCE_MOVE STEPPER=stepper_z DISTANCE=100 VELOCITY=200
This resulted in
Recv: // Klipper state: Shutdown
Issued an M112. Log attached.

You know, I really think the way to handle "move without homing" is to let the user decide. I can see the point of not doing so, but I feel like it's a punishment for those who know what they're doing. The worst part of it is that I have to issue a long, rather arcane command, instead of a GCode. This behaviour alone makes me wonder if I should bother with Klipper, even though it seems like a fantastic program that I REALLY want to use.

If I was a better programmer, and was better at understanding code I did not write, I'd "fix" this myself.

Thank you for making Klipper freely available.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2431260/klippy.zip)

dushyantahuja, 2018-09-30:
I agree and would like to chime in on this. You could have a config
parameter to allow this. Keep it off by default, but allow advanced users
to change it.

Once in a while I need to just move the head up by a few mm, when I'm
cleaning the bed, but cannot. I need to go through the whole homing
sequence, which on my printer is extremely long as I have a belted Z axis
that moves very very slowly. Doesn't matter while printing, but homing from
750mm takes a while.

lar3ry, 2018-09-30:
> `SET_KINEMATICS_POSITION`

`SET_KINEMATIC_POSITION` actually, but yes, that does the trick. I can decide later whether I want to put it in as part of connecting to the printer or using a custom button.

I did not know the FORCE_MOVE did not do acceleration/deaccelleration. That's why my second command had an F20.

I had a look through the docs, looking for SET_KINEMATIC_POSITION (and SET_KINEMATICS_POSITION), but could not fnd any reference to it. Nor could I find anything in the downloaded klipper-master. Yet I am pretty sure I saw something about it somewhere. Perhaps it was mentioned on Faceplant.

Anyway, thanks for the info.

Larry

KevinOConnor, 2018-09-30:
FYI - https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#force-movement

lar3ry, 2018-10-01:
> 
> 
> FYI - https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#force-movement

Thanks for the link.  The copy of the klipper-master I downloaded had a different G-Codes.md file, and did not have the SET_KINEMATIC_POSITION documentation.

lar3ry, 2018-10-03:
Solved by using SET_KINEMATIC_POSITION

closed 2018-10-03 by lar3ry