Klipper3d/klipper #712: [FR] Use max PWM instead of PID during heatup

GarrethX, 2018-09-30:
Would it be possible to have an option to use full heater power while the temperature is below/above user-set range value during heatup, instead of PID? 

After PID autotune, I've got pretty good results when it comes to temperature stability, however I noticed that during bed heatup I can hear power supply fan turning on and off, normally it means that table is not using full power that it could get. 

In marlin for example that feature is defined in config as:
`#define PID_FUNCTIONAL_RANGE 6 // If the temperature difference between the target temperature and the actual temperature
                                  // is more than PID_FUNCTIONAL_RANGE then the PID will be shut off and the heater will be set to min/max.`

This allows for a better fine tuning PID controller, while still maintaining best heatup time.

Thanks!

KevinOConnor, 2018-10-01:
The PID does always turns fully on at start.  The log will show the pwm output values for all the heaters.  If you're experiencing a problem you should attach the full log as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

GarrethX, 2018-10-01:
You're right, however I had to play with PID parameters by hand a little bit to get things better.

After auto tune with 200x300mm varipower bed from e3d I'm getting the results like in the attached log. 
At stamp 108.7 you can see that PID regulation kicks in, however with that big bed we are still 7 degrees off the target. The target 60 degrees are reached at stamp 172.8 (stable state is achieved little bit further as it overshoot by 1 deg)

Of course in this situation I could go with higher P values, or some play with I or D, however with suggested functional range feature (set to e.g. +/- 2-3 degrees) maybe I could achieve bed heatup much faster. 
At the same time I could actually make PID less aggressive to achieve better stability as reaction time at my nicely isolated heat bed doesn't matter this much,

[klippy.log](https://github.com/KevinOConnor/klipper/files/2435228/klippy.log)

That's just my example, I don't want to be the guy saying "let's copy marlin", however that functionality could have some uses with systems that are a little bit harder to stabilize and keeping shortest possible time of heatup. I can live without it however, as other features of klipper are great:)

KevinOConnor, 2018-10-16:
In general, I prefer the simpler PID than the PID implementations with lots of tuning parameters.  My experience is that they add more complexity than benefit.

I did spend a bunch of time earlier this year in seeing if it was possible to use a "first order plus dead time" model of the heater to improve the heater response.  (It's on the work-fopdt-20180405 branch.)  With that code I found that I could get my heaters to perform the initial heat up with zero overshoot and zero undershoot.  Unfortunately, that code did not react well to disturbances and was unable to maintain a consistent temperature during printing.  That said, I personally am more interested in developing better models of the heater than on tweaks to the pid algorithm.

-Kevin

closed 2018-10-25 by KevinOConnor

hg42, 2018-11-02:
what @GarrethX said, is exactly what I also reasoned about on my pid_settle_xxx PR.

There are two cases of temperature regulation.
One strategy is to stay at a temperature while resisting every outside influence.
The other strategy is to follow a commanded change in temperature as fast as possible.
It's totally obvious, that these two very different goals need very different PID parameters.

In my case the strategy mismatch means a 30 min heatup, and that with a 2000W heater that is very capable of heating up in a few minutes. If I tune PID to heatup in for example 10 mins, it's overshooting all the time during the print. There is simply no solution for both startegies at the same time with the same PID parameters.

You once said simple PID works for most...well, it's only because most bed heaters are very weak and don't need much regulation, because they are designed to be at their limits all the time.
That's also the only reason why simple switching on and off also works for some heaters. Do that with my heater and you get a fire.

Ideally, there would be two parameter sets, one for a *stable* temperature while printing and one for the a commanded *change* of temperature.
The "functional range" method with full on/off of the heater is a simplification of the second parameter set which should work in most cases (not sure if it would completely work for my bed and heater, but at least it would cut down the heatup phase to a minimum).

GarrethX, 2018-11-02:
I was the only one in the discussion, and nobody else seemed interested:)

More complex control algorithms are not always better. They are much harder to tune or adjust. Even in industry applications I was using PID algorithm for stabilization parameters like airflow or temperature very successfully. 

Double or even triple PID is broadly used, however in our application we just need a P for initial heatup and PID for stabilization. To be more "on the edge" we could have different PID for situation where we've got print fan ON, or adjust P part in relation to extrusion speeds, but that seem to be such too big overkill:) 

What helped in my case is reducing the smoothing time. I didn't even know that we could configure that, and I don't know why we use temperature smoothing in the first place.. I mean graph looks nice, but PID is messed up when smoothing is too long like in default value.
Also maybe PID chain could be a little bit faster. I'm not sure how far are we using the advantages of klipper work offload to RPI in terms of speed of temperature gathering, processing, and outputting PID.

hg42, 2018-11-02:
I was offline (from 3D-printing) for a while, today I catched up with all the mails and issues...otherwise I would have dropped in earlier...
According to the second PID parameter set, I tried to look from a more general point of view.
You are right, there could be even more cases...(fan etc.). Fortunately these are usually in the reach of the main parameter set.

That said, I could create my own heat up sequence by applying another set of PID parameters. Because of my special bed/heater combination I am already using a three step approach. I first heat up the bed to some temperature below the target. Then I heat up the hotend and then I set the bed to the target temperature.
My only problem is the long settling time, because of the default limits. This forces me to manually control the whole procedure, which is a pity.