Klipper3d/klipper #713: Timeout on Serial connect

mcarpenterjr, 2018-09-30:
Keep getting hit with timeout issues. OctoPrint recognizes it is connected to the printer and confirms in the terminal
```
Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x7efe5c357590, open=True>(port='/tmp/printer', baudrate=57600, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.6.0-459-gb112cac FIRMWARE_NAME:Klipper
Send: N2 M21*18
Recv: ok
Send: M115
Recv: ok FIRMWARE_VERSION:v0.6.0-459-gb112cac FIRMWARE_NAME:Klipper
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M112
Send: N3 M112*34
Recv: ok
Send: N4 M104 T0 S0*37
Recv: ok
Send: N5 M140 S0*96
Recv: // Shutdown due to M112 command
Closing serial port due to emergency stop M112.
Recv: // Once the underlying issue is corrected, use the
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline (Error: Closing serial port due to emergency stop M112.)"
```

The klippy.log, shows otherwise. When sending commands like homing the machine nothing happens.

```
Starting Klippy...
Args: ['/home/sysadmin/klipper/klippy/klippy.py', '/home/sysadmin/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.6.0-459-gb112cac'
CPU: 2 core Intel(R) Pentium(R) D CPU 3.00GHz
Python: '2.7.15rc1 (default, Apr 15 2018, 21:51:34) \n[GCC 7.3.0]'
Start printer at Sun Sep 30 13:24:14 2018 (1538328254.8 705697.2)
===== Config file =====
[stepper_x]
step_pin = PD7
dir_pin = PC5
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC2
position_endstop = -15
position_max = 220
position_min = -15
homing_speed = 40

[stepper_y]
step_pin = PC6
dir_pin = PC7
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC3
position_endstop = -8
position_min = -8
position_max = 220
homing_speed = 40

[stepper_z]
step_pin = PB3
[... 330 lines cut ...]
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Shutdown due to M112 command
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Shutdown due to M112 command
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Timeout on serial connect
Timeout on serial connect
```

The server that the printer is connected to was recently upgraded to Ubuntu 18.04 and everything broke. I had to reinstall and configure OctoPrint and Klipper. I probably should not have but I also reflashed the firmware on the printer as well. Also is it normal to have message duplication in the log file?

swatti87, 2018-10-01:
Preeetty much same issue here. Fresh SD-card format and fresh install of klipper. 
Used Ramps config and set pins etc correctly.

Octoprint console/terminal "status" just says printer not ready

Klippy-log full of timeouts and try again later messages.

KevinOConnor, 2018-10-01:
@mcarpenterjr - there's been several reports of issues with Ubuntu 18.04 - see issue #443.  What type of micro-controller are you using?

-Kevin

mcarpenterjr, 2018-10-01:
@KevinOConnor Stock Anet A8 Board, so That falls into the Atmega 1284 family, I believe.

mcarpenterjr, 2018-10-04:
Hey is it possible to flash from a VM?

mcarpenterjr, 2018-10-06:
`Connected to host via /tmp/printer @0bps` What/where does the @BPS come from?

KevinOConnor, 2018-10-08:
I'm going to close this in favour of the existing issue #443.  Short term workaround would be to compile the firmware on something other than Ubuntu 18.04.

-Kevin

closed 2018-10-08 by KevinOConnor