Klipper3d/klipper #715: [Feature Request]

lar3ry, 2018-10-01:
I have noticed that I can't send more than one command per line. When I try to make a command button with multiole GCode commands in it, separated by colons, I get an error. It complains about the colons. Is the parser not taking them as being the same thing as a newline?

FHeilmann, 2018-10-01:
try separating them by '\n' instead of colons/semicolons

i.e.
`G28 X0 Y0 \n G0 X0 Y0 \n G0 Z5`

lar3ry, 2018-10-01:
> try separating them by '\n' instead of colons/semicolons
> i.e.
> `G28 X0 Y0 \n G0 X0 Y0 \n G0 Z5`

Typed in...
g28 \n g1 Z5 \n G1 x100 Y0

Result...
Send: G28 \N G1 Z5 \N G1 X100 Y0
Recv: // Unknown command:"G28 \"

lar3ry, 2018-10-03:
Solved by putting macros into the config file, and calling them by name in the custom uttons.

closed 2018-10-03 by lar3ry