Klipper3d/klipper #718: Extrude without hotend heat fails

lar3ry, 2018-10-01:
Trying to extrude while cold, in order to calibrate extruder.
Minimum temperature set to 0.
Send: FORCE_MOVE STEPPER=extruder DISTANCE=5 VELOCITY=2
Recv: ok
Recv: // Klipper state: Shutdown

[klippy.log](https://github.com/KevinOConnor/klipper/files/2434812/klippy.log)

jakep82, 2018-10-01:
You're doing this all wrong.  You need to set min_extrude_temp in your config, and then just use a regular G1 Exxx Fxxx command.

lar3ry, 2018-10-01:
First thing I tried...

Send: G1 E5
Recv: // Extrude below minimum temp
Recv: // See the 'min_extrude_temp' config option for details
Recv: !! Extrude below minimum temp
Recv: ok

jakep82, 2018-10-01:
Please re-read what I wrote, and what the error you just posted says.  Min_temp is not the same as min_**extrude**_temp.  If you don't explicitly set min_extrude_temp, it will default to 170 as described in the example.cfg file.

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg

lar3ry, 2018-10-01:
Ahh... my bad. Thanks.

However, after making that change, I get this:
Send: G28
Recv: // Failed automated reset of MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Failed automated reset of MCU 'mcu'

Issued M112

[klippy.log](https://github.com/KevinOConnor/klipper/files/2435558/klippy.log)

jakep82, 2018-10-01:
This is a completely separate problem.  I would recommend closing this issue and starting a new one.

lar3ry, 2018-10-01:
Well, I came back to it, disconnected, reconnected, issued a G28, and it worked fine. I can also run the extruder without heating. No idea what happened, as I did absolutely nothing with it since the above happened.

closed 2018-10-01 by lar3ry

jakep82, 2018-10-01:
Glad you got it working.

theopensourcerer, 2018-10-02:
@lar3ry If you found things frustrating that suggests the documentation wasn't clear or accessible enough. It would be great if you could identify areas that could be improved.

lar3ry, 2018-10-02:
Some of it was a matter of having to go through hoops to do simple things that are done on other boards with existing GCodes, and some was a result of not having examples explained clearly.

Example 1: I use a RUMBA board on a Reprap style printer. I was able to use a GCode to start a fan and turn on lights.  All I had to do was to issue "M42 P8 S255". Reading the example.cfg, it looked like what I had to do was to use a "SET_PIN" code. Fine, but first I had to find the pin by name. I had nothing to go on except the RUMBA example config file, which used some naming scheme I had not seen before, and I was not willing to try changing to pin numbers, because I was not sure how to translate them.
I finally found a file that, with a lot of head scratching, allowed me to find out the pin I wanted to toggle. During all this, I kept getting errors that shut things down.

Example 2: I wanted to be able to move the steppers before homing, which, IMHO, is essential to make sure everything is moving in the right direction before committing to a (possibly disastrous) G28. I spotted the "FORCE_MOVE" command, which sort of worked, causing a lot of complaining from the steppers I was moving. As well, every time I used it, Klipper would terminate on the next usage.  I had not seen the "SET_KINEMATIC_POSITION" as relevant, because there really was no indication that it actually allowed moves before homing. It sounded to me like the sort of thing you would use after homing, then moving to an X,Y, and Z position, then telling it that it was at 0,0,0, such as might be done when using the mechanism to do circuit  board etching, vinyl cutting, or pen plotting.

Example 3: Seems I could not use multiple GCodes from the terminal text box, or in controls configured using "Custom Control Editor" or "Terminal Commands" plugins. I solved that by finally making macros specific to the control I wished to use, then calling them with the name of the macro. Problem solved, but it took a long time and a lot of trial and error.

Currently, I am working on homing. I have been trying to avoid having to issue commands to raise the Z axis a few mm before homing X and Y, then homing Z, because sometimes the nozzle is close enough to the bed to occasionally contact it while doing X or Y motions. There are a lot of "interesting" comments in the "homing_override" section that I am trying to fathom.

BTW, while I have you on the line, when homing  is there a particular reason for the long delay between the first endstop contact and the raise/lower second motion?

jakep82, 2018-10-02:
Did you read the config check instructions?  It's linked at the end of the main installation instructions. 
 The stepper_buzz feature does exactly what you want with regard to verifying stepper direction.

https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

For the homing override, I think you just need something like the following:

> [homing_override]
set_position_z: 0
gcode:
    G90
    G0 Z5 F300
    G28