Klipper3d/klipper #721: Support for rpi serial connection via gpio to einsy rambo atmega2560

simonlee, 2018-10-01:
I have great success using klipper on an rpi3 and octopi connected to an einsy rambo (one used for the prusa mk3) via the USB interface. 

It is possible to use the hardware UART on the rpi as the serial interface to connect to the einsy by connecting gpio 14 and 15 (and the one for the nRESET?) to the appropriate pins on the einsy's j19 pins:
[image]

However, the klipper firmware needs to be aware of this. Could someone with more knowledge nudge me toward the right direction on getting this implemented? Any information would help.

I don't care about being able to flash it. Just have it be able to connect to the mcu via rpi gpio serial interface.

jakep82, 2018-10-01:
I have my Pi connected to my printer via UART without issue.  My MCU is an Arduino Due, but it should work the same.  I had to disable the serial terminal on my Pi in order to connect to the main UART.  See the link for more info.  It's easy to do with raspi-config.  I connect to it with /dev/serial0 in my printer.cfg.

https://www.raspberrypi.org/documentation/configuration/uart.md

simonlee, 2018-10-01:
I don't think it should work the same.

I have already disabled the bluetooth that takes over the hardware UART by using the pi3-disable-bt overlay. After doing that, I can see that /dev/serial0 points to /dev/ttyAMA0.

jakep82, 2018-10-01:
Did you disable the modem services?

> sudo systemctl disable hciuart <

All I did was disable the serial terminal, wire Tx to Rx and vice versa, and edit my printer.cfg to connect to /dev/serial0.  Works perfect for me, but I've never connected to an ATMEGA.

jakep82, 2018-10-01:
I think I also disabled BT, but it's been quite a while since I did it so my memory is a bit fuzzy.  I'm fairly certain you still need to disable the serial terminal.

simonlee, 2018-10-01:
Yes I had already disabled hciuart service and also disabled the serial console in the /boot/cmdline.txt.

jakep82, 2018-10-01:
So then what is the question?  Just change the serial connection in printer.cfg to /dev/serial0 and you're done.

`[mcu]
serial: /dev/serial0`

simonlee, 2018-10-01:
Sorry if I wasn't clear. I've already tried that. I get a serial timeout when using /dev/serial0 or without using that alias and directly using /dev/ttyAMA0.

When I had the prusa firmware loaded though, I was able to use /dev/ttyAMA0. So there must be some kind of difference in the prusa firmware that allows it to work but not in klipper.

jakep82, 2018-10-01:
What baud rate did you use when you flashed the MCU?  A log file might help.

simonlee, 2018-10-01:
I've tried both 250000 and 115200 when flashing, with the corresponding baud rate set in the printer.cfg. Here is a log showing the timeouts:
```
[mcu]
serial = /dev/serial0
baud = 115200

=======================
bed_mesh: generated points
(10.0, 10.0)
(100.0, 10.0)
(190.0, 10.0)
(190.0, 100.0)
(100.0, 100.0)
(10.0, 100.0)
(10.0, 190.0)
(100.0, 190.0)
(190.0, 190.0)
Extruder max_extrude_ratio=20.787584
Starting serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
```

simonlee, 2018-10-01:
My guess, as I've stated initially, is that the klipper firmware is unaware of the particulars of the einsy rambo board's serial interface. If anyone has gotten their prusa mk3 working with klipper via serial gpio, please share how you did it!

KevinOConnor, 2018-10-02:
Klipper uses the RX0/TX0 pins by default.  You'd need to select UART1 to use the RX1/TX1 pins.  I just recently added support for selecting alternate UARTs on the 2560 (commit d348f0da).

-Kevin

simonlee, 2018-10-02:
Excellent! I've confirmed that this works now. Thank you Kevin for the prompt fix and overall amazing work.

closed 2018-10-02 by simonlee