Klipper3d/klipper #723: z-offset

povlhp, 2018-10-02:
I have an Ender-3, controleld by Octoprint on OrangePi Lite.
Was running Marlin 2.0.x

My problem is, that on first print, the nozzle burried itself in the bed.

There are 2 problems, one is in 
`printer-creality-ender3-2018.cfg`

`# If you don't like the 0.5 you should change this to 0.0
position_endstop: 0.5`

I think it is better to fail on the safe size. So saying the endstop is at 0.5 as default is BAD.

I know there is a question on this in the FAQ.md. But I am not sure I agree. It should be as seamless as possible to move from Marlin to Klipper to get more followers. Having something as destructive as burrying the nozzle in the bed is bad. If you want to keep this, it needs to be emphasized in the Installation.md  documention.

Documentation should also refer to the example config documents more loudly, as they explain all the options. Most people will likely do like me, and just copy a printer config and think all is well.

The other issue is, that even though I changed endstop to 0.0, I still was way too close to bed. Had to add Z-offset of 0.3 in Cura to get correct 1st layer adhesion compared to Marlin.

jakep82, 2018-10-02:
It seems odd to me to home the Z axis below the surface of the bed, but if this is what the Ender 3 does you can fix the Z offset issue with printer settings rather than using an offset in your slicer.  Just add a position_min setting of -0.3 in the stepper_z section, and also change position_endstop to -0.3.

povlhp, 2018-10-02:
Also not sure what is going on. Homing will go to the right place. 
But then for printing it will move to start position, Z = 0.5 and issue M82 to define this as 0.2 (doing the extra offset). Still more than close enough. 

Will try to remove all offset and calibrate it again (still by hand). Then try Marlin again.

BlackStump, 2018-10-03:
I have a Ender3 and  I have not experienced any of the issues you mentioned with the Z endstop position at 0.5. The default is still position_min 0
When you level the bed with the big adjusting wheels on each corner you still send the gcode in the octoprint terminal to ie X10 Y10 Z0  for the front left corner and adjust that position to a paper thickness between the nozzle and the bed as per Ender3 instructions when you first commissioned the printer ? then do the other 3 corners
If you and levelling the bed by only using X and Y coord you will be at Z0.5 if that makes sense.
In anycase if you have changed to position_endstop: 0 the procedure to level the bed is exactly the same.
For what I am reading you need to level the bed correctly

povlhp, 2018-10-03:
I re-levelled my bed, did not go back to Marlin. And it works.

But basicly, the issue with assuming the endstop is 0.5mm up by default is, that the nozzle will move down 0.2-0.3mm typical (depending on first layer height) to reach the 0.3 or 0.2mm first layer. 

If you have calibrated your bed on Marlin, your nozzle will be 0.1mm (1 paper thickness) above the bed, and moving it 0.2mm or 0.3mm down on first layer will jam the nozzle into the bed.

So assuming endstop is at 0.5mm and there is plenty of room below is an unsafe default. Should be noted more clearly in docs. That is all I ask for.

Bill26, 2018-10-10:
Hi, i had the same issue. 
To fix it I had to change the z_offset to 1.7 for my bed levelling config and now it perfect. I'am using Bltouch on Ender 3

KevinOConnor, 2018-10-16:
I'm going to close this for now.  If there's a better setting for the printer than please submit a pull request with the change.

-Kevin

closed 2018-10-16 by KevinOConnor