Klipper3d/klipper #724: CR-10S 4 Config

IanCoburn, 2018-10-02:
I'm really excited about installing Klipper, and I've arrived at the point of testing out a print.  I have a CR-10S 4, which is a larger scale version of the CR-10S.  I'm using the Cr-10S config to get off the ground, and as you might expect, there are issues using that to drive the CR-10S 4.  Namely, the heated bed is different, and can't properly be controlled, saying that it hits a critical error and can't continue heating the bed.  

My question is how do I get the values I need to use in my config file for my heated bed?  I've done a number of searches, and it appears that there isn't anyone else using my model, at least that have posted about it.  Is there a way to tell by labels on the heated bed itself that I should be looking at?  Any help here would be awesome, as I'd love to start using the software.

ruevs, 2018-10-03:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

IanCoburn, 2018-10-05:
Thanks Kevin, and thanks for the software.  I assumed that was what ruevs' post meant, but just pasting a link didn't really mean much or help me out.  I'll take a look and see what I can divine from the various sources you mention.  I appreciate it, and if I post again, I'll include a log.

Maybe there could be an FAQ that could answer that question, among others?

closed 2018-10-05 by IanCoburn

IanCoburn, 2018-10-10:
I went through some tests again, and found the bed to heat improperly over time.  The nozzle doesn't heat up at all.  Based on the error, I scoped out the  verify_heater heater_config_name in example-extras.config, and it is unclear how I might adjust these settings to work as expected.  The simplest explanation in my mind is that they don't have the correct settings in the config to suit the CR-10 S4, for both the heated bed and the nozzle-- the printer-creality-cr10s-2017.cfg values obviously not carrying over to the larger model with a larger bed size.  Barring anything obvious found in the logs, I'll start trying to ascertain the values from the original firmware.

Here's the settings I'm using, from printer-creality-cr10s-2017.cfg:

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: pid
pid_Kp: 690.34
pid_Ki: 111.47
pid_Kd: 1068.83
min_temp: 0
max_temp: 130

Here's my log:

[klippy.log](https://github.com/KevinOConnor/klipper/files/2465387/klippy.log)

jakep82, 2018-10-10:
You need to determine the correct heater pins for your bed and hotend.  I did some quick Googling and had trouble finding anything definitive.  If you have a version of Marlin that works on the board, you can find the correct pins in there.

IanCoburn, 2018-10-20:
So after an incredible amount of testing, I ultimately realized that I can get the heated bed up to temp by increasing the temp slowly over time, specifically in 5 degree increments.  If the jumps are any larger, I get the same issue indicating that the bed is heating up at the improper rate  CR-10 S4 beds are notoriously slow to heat up-- could that be the issue?

I can set my nozzle to temp (200) without any issues, and it heats up very quickly.

I'm attaching my most recent log wherein I try to print the test square (having inched my way to the proper bed temp).  During the middle of it, when the print fails, I try to cancel the job.  At that point Klipper stops responding, and also suddenly says the nozzle isn't heating at the proper rate.

Any other ideas?  I'm just about to switch to Marlin to see if its not just the firmware, but perhaps my printer as well.  

[klippy.log](https://github.com/KevinOConnor/klipper/files/2497775/klippy.log)

ruevs, 2018-10-22:
Perhaps adding such settings would help:
https://github.com/KevinOConnor/klipper/commit/c7f0d039ac562d2d6d3ef8876faffdde48a8bf89#diff-ad28d0cdd9e65c147782fa68b19035aaR65