Klipper3d/klipper #725: Instrument Klipper to expose metrics

cassianoleal, 2018-10-02:
This is more of a feature request / conversation starter.

It would be very interesting to get certain metrics out of Klipper. Initially I'm thinking things like:
- a counter for number of instructions sent over the serial (so we could get rates, totals and other goodies - potentially cross-reference with things like buffer queues etc to help diagnose issues with starvation).
- temperatures (bed, extruder(s))
- process data (memory used, threads, whatever other things make sense)
- other metrics?

Ideally for me it would be exposed in Prometheus format.

I already run `node_exporter` on the Pi's and can get a lot for free there but nothing specific to the print jobs and the printers. Prometheus scrapes them and I can create nice dashboards in Grafana.

jakep82, 2018-10-02:
Some of this information is already available in the log file.  You can generate a log graph that includes bandwidth, host buffer, and MCU load information.  Temperatures are also in the log file, and I'm sure they can be extracted fairly easily.

https://github.com/KevinOConnor/klipper/blob/master/docs/Debugging.md

FHeilmann, 2018-10-02:
Extras can access pretty much all the metrics you mentioned and export in whatever means python is capable of.

I suggest you look into writing an extra for Klipper.

cassianoleal, 2018-10-02:
@jakep82 I'll see if I can extract that info and expose in a Prometheus format, thanks.

@FHeilmann What do you mean by extras?

cassianoleal, 2018-10-02:
It looks like I could use the [Grok exporter](https://github.com/fstab/grok_exporter) to parse the log and expose metrics.

FHeilmann, 2018-10-03:
Klipper has a modular addon system (called extras) which allows to easily write addons in python that can interact with the klipper code. Check /klippy/extras/ for the existing ones.

aeropic, 2018-10-07:
the octoklipper module offers a performance graph feature that displays:
-MCU loads
- bandwith
- buffer
- ...

cassianoleal, 2018-10-07:
I know. I want to export those metrics so I can add them to my Pi's dashboards with CPU, I/O, networking, load, etc metrics, then I can post annotations to it and make correlations, have historical data, etc.

KevinOConnor, 2018-10-16:
It seems the original question was answered, so I'm going to close this.

closed 2018-10-16 by KevinOConnor