Klipper3d/klipper #727: M112 not aborting right away

povlhp, 2018-10-02:
When I send M112 during pre-heat phase, it will not abort before the temperature is reached.
Can we please have the M112 take precedence over whatever waiting we are currently doing.

Should be minor thing. Often I start preheat, then thunks about something to change in slicer, and goes back to it.

KevinOConnor, 2018-10-04:
Klipper will react immediately to an M112 command.  If you're having an issue you need to attach the Klipper log file: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

jakep82, 2018-10-04:
I've issued an M112 during pre-heat on several occasions, and Klipper has always aborted immediately.

povlhp, 2018-10-05:
I will do more tests, and include the log. I experienced it twice.

KevinOConnor, 2018-10-14:
Okay - closing this for not.  I'll reopen if you have a log from the event.

-Kevin

closed 2018-10-14 by KevinOConnor