Klipper3d/klipper #731: Feature req: retraction free pressure advance

povlhp, 2018-10-03:
The one thing that stops me using pressure advance on every attempt is the noise of constant retraction taking place. I have a geared extruder with 500 e-steps/mm.

I suggest that another method is developed, which will not cause any reversing of filament, but rather just use the oozing to obtain the same effect.

We know that the built-up pressure in the nozzle/hotend increases to a certain point by pushing filament thru, and decreses by oozing/depositing filament. It should be possible to find a formula describing the oozing per second, and one to describe the pressure build-up.

Cura already has a coasting setting to do something along these lines, but I imagine that Klipper could do a better job. And Cura does not look at pressure build-up after travel, which is important as well. I want slower acceleration until the flow can keep up. And it sems Cura does coasting at constant speed, where Klipper could handle it by adjusting speed better to the pressure drop formula. I guess pressure could drop with the square of time.

Maybe the existing algorithm already do that, but the retractions are killing it for me. Then rather slow down a bit.

KevinOConnor, 2018-10-04:
I suspect you'll need to implement this if you want to test it.  I suspect it will not work well - the nozzle pressure is released over an exponential time - if you want to avoid oozing without retraction it will likely take 5+ seconds after every extrusion move before oozing stops.

Separately, if you're hearing loud noises from the extruder that indicates the printer acceleration is too high for the configured pressure advance.  Reducing acceleration often helps - there's some details on this at: https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md

povlhp, 2018-10-05:
Not much Python experience, just minor changes in a few projects. But maybe I can come up with something.
What I was expecting was something closer to coasting. I know that the nozzle leaks forever, and I was not planning for it to give string-free travels, rather for it to be a simple solution to give better start/end of lines. Smaller retract on travels.
Will try to reduce acceleration. I do have a belted extruder, so 500 e-steps. So I have a much larger piece of hardware that is moved (a 16-GT2 pulling a 60-GT2 down to a knobbed shaft, so 44mm diameter wheel pulling 8mm teethed shaft).  But in Marlin, normal retract at 60mm/s works fine (500mm/s^2 acceleration for retract,). Had the same noise issues in Marlin. Forward acceleration is only 100mm/s^2 though.

povlhp, 2018-10-05:
Another comment, if extruder acceleration is limited by max XY acceleration, and I run 3000 mm/s^2 on X+Y, then extruder should be limited to outputting (assuming 0.4mm 1mm in = 15mm out)  sqrt((3000+3000)^2) / 15 = 400mm/s^2. Or am I off here somehow ?

the example config says
#max_extrude_only_accel:
#   Maximum acceleration (in mm/s^2) of the extruder motor for extrude
#   only moves. If this is not specified then it is calculated to
#   match the limit an XY printing move with a
#   max_extrude_cross_section extrusion would have.

max_extrude_cross_section would be 4 * 0.4^2 = 0.16mm^2
Input is 2.4mm^2

So in to out is 1 : 15 (this is the 15 I used above).

Is extrude and retract acceleration the same ?

KevinOConnor, 2018-10-08:
The max_extrude_only_accel and max_extrude_only_velocity only apply to moves that are "extrude only" - they do not apply to any normal XY move with extrusion.  (Technically, they also apply to any retraction move, even if the head happens to have an XY movement during a retraction.)

There are no secondary limits that constrain the pressure_advance setting - the user is expected to set pressure_advance correctly.  If pressure_advance is set large, then the extruder will be commanded to perform large velocity shifts.

-Kevin

closed 2018-10-16 by KevinOConnor