Klipper3d/klipper #732: REQ: TMC2130 example config & help

swatti87, 2018-10-03:
This coding-stuff is spooky. I'm not very good at this but im learning... Slooowly.

I had a "custom" branch, if that is the correct way to call it, of Klipper, made by local haxor-wizards who are also behind a custom control board "Oulware" that is able to use the TMC2130s without any extra wiring. 
( http://www.oulware.org/index.php/Printer_Controller )

Now i'm upgrading to the official Klipper (for auto bed-leveling) but there are prooooblems.  Mostly the sensorless homing of the TMC2130

The previous config-file is not directly usable with the official Klipper and has new or different names in some... Things.

So, what i could really use is a "working" config-file from someone.
I can copy the needed lines from the config and leave anything un-needed out, like the comments on the example-file here on github. 

I should be able to to copy-paste almost all of the pins and values from my old config, apart from maybe couple: 
cs_pin
Diag1_pin

I have no idea what those are, or where, or how to find em. Most likely i left out something important, ask.

jakep82, 2018-10-03:
query_endstops

https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

See the TMC-2130 section here to configure your drivers.  The SGT value for tuning the endstops will vary significantly depending on what voltage your using for your steppers.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg

I'm also using an OULWare board.  You will likely need to change some pins depending on what driver slots you're using on the board, but here's my printer.cfg as an example.
[OULWare.txt](https://github.com/KevinOConnor/klipper/files/2442571/OULWare.txt)

jakep82, 2018-10-03:
Also you can use this config file for reference.  All of the pins are listed at the top.  The CS pins for each driver are listed, and the diag1 pin is just the endstop pin.  

https://github.com/OULWare/klipper/blob/master/config/oulware.v1.0.cfg

swatti87, 2018-10-04:
Will my endstops show triggered if I move the carrier and open if I don't?

Or how do "physically" trigger them or can I since they are sensorless.

Plan B: is setting SGT too low so it will trigger from any movement and add return movement when triggered. Then start testing what pin causes the return movement...

jakep82, 2018-10-04:
If you use the pin definitions I provided and you have the jumpers installed in the right spots (Xmin and Ymin) it should work correctly.  If you run endstop_query it should show your endstops as open.  Start with a low SGT value of 1 or 2, and work your way up.  You also need to use a reasonably fast homing speed.  I'm not sure what the minimum speed is, but I've had good success at 50mm/s.  Also note that the minimum SGT value will change when the drivers are hot.  I can home with an SGT of 2 when my drivers are cold, but they trigger early at that value when they're hot.  I use 4 which works well regardless of the driver temperature.

swatti87, 2018-10-04:
Gotta say, i would have NEVER gotten this to run without your config as refecence... I cant thank you enough.

Ok, things run preeeetty well now.

Last step: Is there a way to disable the home-before-move-anything feature for the Z-axis? I'm okey if X and Y stay with it on.
I have no Z-sensor installed at the moment, i calibrate it by hand untill i get BLtouch running again... Aaand i need mount for it printed.

jakep82, 2018-10-04:
You can use set_kinematic_position to do this. I believe you need to enable it in your config. See the link for info. 

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

theopensourcerer, 2018-10-05:
You don't need the square brackets... Those are commonly used to indicate that the information within the brackets is optional. This is also explained the documentation:

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#force-movement

I am guessing the bad parameter formatting is why you get the message. I agree it could be worded better - more like:

// Unknown parameter for command "SET_KINEMATIC_POSITION" 

or something like that perhaps.

swatti87, 2018-10-05:
I had to enable the "force-move" thingy in the config first. Devious.

swatti87, 2018-10-05:
How do i stop this thing from sending "bare" G28 at start... Its not in the start scripts. Since my Z wont home.

jakep82, 2018-10-05:
Klipper doesn't issue any g-codes.  If you were getting a plain G28, it was coming from your slicer.  If you can't figure out how to prevent your slicer from adding the G28, then just edit the g-code file with a text editor and delete the G28 before uploading to Octoprint.

I have a BLTouch, but I've never used it with Klipper so I can't help with setup.  I would suggest Googling the BLTouch.   It's not hard to find wiring diagrams that show what each wire does.

swatti87, 2018-10-05:
Oookkey...

Should have most things set but where to find a PWM-pin in the OulWare board? Which ones are the PWM ones?
I think i can take it from there.

jakep82, 2018-10-05:
[Controller_v11.pdf](https://github.com/KevinOConnor/klipper/files/2450815/Controller_v11.pdf)

swatti87, 2018-10-05:
> [Controller_v11.pdf](https://github.com/KevinOConnor/klipper/files/2450815/Controller_v11.pdf)

Is that the OulWare board schematic? Looks... Odd.

P5 and P6 pins seem PWM... I'd imagine that would work. Just where on the actual board are they. I take it their pin-name is "ar" and number from schem

jakep82, 2018-10-05:
Just match the pins.  Pin 6 from P6 goes to pin 7 on the AUX1 header, and I believe it would be ar2 in the config.

swatti87, 2018-10-05:
Progress!!
BLtouch "lit up" and is blinking the warning. So, its working, in a way and i most likely did not deep-fry the poor thing.

Lets leave the pin-charts out. Scary stuff that.

Use this:
http://www.oulware.org/images/2/2d/Controller_v1.1.jpg

Draw a red dot on a PWM pin and its number.

jakep82, 2018-10-05:
I already gave you the info you need.  Pin 7 on the AUX1 header is ar2.

swatti87, 2018-10-05:
As in second row, as in the pic and the left pin? (no idea what header really means)

jakep82, 2018-10-05:
Look at the schematic I provided.  Find AUX1.  Find pin 7 on AUX1.  It will have a black 2 next to it signifying the arduino pin to which it's connected.  

Note that I have no idea if that pin will work, I can only say that it's a PWM capable pin that's available on the board.

jakep82, 2018-10-06:
I decided to pull out my BLTouch tonight and wire it up.  It's a bit complicated because the BLTouch needs 5V for the servo, but all of the logic on the OULWare board is 3.3V.  Fortunately 5V is available on the static voltage out pins.  What this means is you'll need to use jumper wires.  For the sake of simplicity, I would wire the signal wire from the BLTouch to the Xmax endstop signal pin.  In your config you'll use ar59 for the probe pin.  I've attached a snippet of my config as a sample so you can see the correct pin callouts.  The 2 pin connector from the BLTouch goes to the ground and signal pins on Xmin.[OULWare_BLTouch.txt](https://github.com/KevinOConnor/klipper/files/2452648/OULWare_BLTouch.txt)

swatti87, 2018-10-07:
There was a jumper in the Xmin sig and diag pins. No idea why. Removed it and 2-wire gnd and sig now in place.

Cant i simply plug the 3-wire power and gnd to the 5v ports? Or do i need some extra wiring there? And 3-wire signal --> Xmax.

A pic of the Oulware board with wiring done would seal the deal, i can copy what i can see.

swatti87, 2018-10-07:
Small progress: The BLtouch now moves it's rod up and down on command... 

I cant seem to make it detect if its up or down... Where am i supposed to put the 2-wire connector wires now? Virtual pin is set but it says the Z is always open, no matter where i plug the wires...

swatti87, 2018-10-07:
Here is my config-file. Still a bit messy but the Z-stepper and probe settings are there...

[printer.txt](https://github.com/KevinOConnor/klipper/files/2454144/printer.txt)

Maybe i missed something.

swatti87, 2018-10-07:
When i use command "PROBE" the probe cycles its pin in and out and i get an error:

Send: PROBE
Recv: // Failed to home probe: Timeout during endstop homing
Recv: // Make sure to home the printer before probing.

swatti87, 2018-10-08:
I have the 2-pin connector to GND and SIG but it cannot detect if the pin is up or down.

With the M119 it should say open when the pin is down, and triggered when its up. It does not.

THAT is the problem.

swatti87, 2018-10-08:
Tested 
Zmin, Zmax and Xmax pins with a plain wire between their SIG and GND. Ran M119 on the terminal and they all report "open"

Software issue says me. Typo somewhere.

swatti87, 2018-10-08:
I had to put a new BLtouch on order. I have two units, both tested recently by a friend, both worked...

I did manage to use a jumper-cable to confirm the pins do work. Its eighter a typo in the config or the probes are dead.

jakep82, 2018-10-08:
I doubt the BLTouch is faulty.  I believe you need to run the activate g-code( SET_SERVO SERVO=bltouch ANGLE=10, SET_SERVO SERVO=bltouch ANGLE=60), manually trigger it, then run query_probe to see if it's triggered.  I believe this worked for me, and then I ran a G28 Z0 and manually triggered the BLTouch to confirm the configuration was working.

closed 2018-10-16 by KevinOConnor

swatti87, 2018-10-24:
ITS ALIVE!! ITS ALIVE!!

Got the chinaasium probe and it works.

Only problem is, it leaves the pin down after the probing

Code for the probe-up is:
SET_SERVO SERVO=bltouch ANGLE=90

But where should i add that in the config to retract it after the probing?

swatti87, 2018-10-24:
Success! Its working, finally!

[homing_override]
set_position_z: 5
axes: z
gcode:
       G90 
       G1 Z10 F600
       G28 X0 Y0
       G1 X100 Y100 F3600
       G28 Z0
       SET_SERVO SERVO=bltouch ANGLE=90
       G1 Z10 F600