Klipper3d/klipper #733: Cannot do SAVE_Config after DELTA_CALIBRATE

DanilaX1, 2018-10-03:
Sorry, maybe I don’t understand something. But after the SAVE_Config command and all calibration steps. The answer comes but the file printer.sfg does not change
Recv: // stepper_a: position_endstop: 301.433665 angle: 90.879074 arm: 271.500000
Recv: // stepper_b: position_endstop: 303.791815 angle: 210.986618 arm: 271.500000
Recv: // stepper_c: position_endstop: 304.067574 angle: 330.000000 arm: 271.500000
Recv: // delta_radius: 134.777008
Recv: // The SAVE_CONFIG command will update the printer config file
Recv: // with these parameters and restart the printer.
Recv: ok
Recv: // Klipper state: Shutdown
Send: M105
Recv: ok T0: 31.7 /0.0 B: 50.1 /0.0
Send: M105
Recv: ok T0: 31.7 /0.0 B: 50.0 /0.0
Send: M105
Recv: ok T0: 31.6 /0.0 B: 49.9 /0.0
Send: M105
Recv: ok T0: 31.6 /0.0 B: 49.8 /0.0
Send: M105
Recv: ok T0: 31.6 /0.0 B: 49.7 /0.0
Send: SAVE_CONFIG
Recv: // MCU 'mcu' shutdown: No next step
Recv: // This is generally indicative of an intermittent
Recv: // communication failure between micro controller and host.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: No next step

[klippy.log](https://github.com/KevinOConnor/klipper/files/2443176/klippy.log)

KevinOConnor, 2018-10-05:
Unfortunately, it looks like you hit the same problem raised in issue #640 - the host code is so busy performing calibration calculations that it misses an important deadline.  I still need to think on how to fix this.  I suspect this will be a very rare event - if you rerun DELTA_CALIBRATE does it succeed?

-Kevin

DanilaX1, 2018-10-05:
I tried to run DELTA_CALIBRATE several times. After each calibration, I run SAVE_Config several times. but the config was not updated. I entered the calibration result in the manual.

wakeboy180, 2018-10-08:
I have the same issue with my system.

DanilaX1, 2018-10-08:
How can I miss this error?
I have already calibrated in manual mode. Config corrected in manual. Data received from the terminal
I want to do an extended calibration.
 
Send: DELTA_ANALYZE CENTER_DISTS=72.9,72.3,72.85,73.1,72.6,72.96
Recv: ok
[...]
Send: DELTA_ANALYZE OUTER_DISTS=72.9,73,72.05,73.1,72.9,72.6
Recv: ok
[...]
Send: DELTA_ANALYZE OUTER_DISTS=72.9,73,72.05,73.1,72.9,72.6
Recv: ok
[...]
Send: DELTA_ANALYZE CENTER_PILLAR_WIDTHS=9.1,9,8.9
Recv: ok
[...]
Send: DELTA_ANALYZE OUTER_PILLAR_WIDTHS=9.05,9.2,8.8,9,9,9
Recv: ok
[...]
Send: DELTA_ANALYZE SCALE=1.0
Recv: ok
[...]
Send: DELTA_ANALYZE CALIBRATE=extended
Recv: !! Must run basic calibration with DELTA_CALIBRATE first
Recv: ok
[...]

Parameters after manual calibration:
[Parameters after manual calibration.txt](https://github.com/KevinOConnor/klipper/files/2457301/Parameters.after.manual.calibration.txt)

DanilaX1, 2018-10-08:
> I have the same issue with my system.

What type of system do you use?
I installed OctoPrint + Klipper on an old netbook on atom processor. The operating system is clean last lubuntu .

wakeboy180, 2018-10-08:
Raspberry Pi

DanilaX1, 2018-10-09:
How can I manually set last_probe_positions ?

KevinOConnor, 2018-10-09:
The DELTA_ANALYZE command will only work if one is able to run DELTA_CALIBRATE, SAVE_CONFIG, and print the calibration object with the settings applied from DELTA_CALIBRATE/SAVE_CONFIG.  It's not easy to manually update the config so that DELTA_ANALYZE works, but if you want to try, you can extract the config values from the previous log:
```
save_config: set [printer] delta_radius = 134.777008
save_config: set [stepper_a] angle = 90.879074
save_config: set [stepper_a] arm_length = 271.500000
save_config: set [stepper_a] position_endstop = 301.433665
save_config: set [stepper_b] angle = 210.986618
save_config: set [stepper_b] arm_length = 271.500000
save_config: set [stepper_b] position_endstop = 303.791815
save_config: set [stepper_c] angle = 330.000000
save_config: set [stepper_c] arm_length = 271.500000
save_config: set [stepper_c] position_endstop = 304.067574
save_config: set [delta_calibrate] height0 = 0.0
save_config: set [delta_calibrate] height0_pos = 24124.000,24294.000,24324.000
save_config: set [delta_calibrate] height1 = 0.0
save_config: set [delta_calibrate] height1_pos = 21353.000,29553.000,29517.000
save_config: set [delta_calibrate] height2 = 0.0
save_config: set [delta_calibrate] height2_pos = 23541.000,23787.000,32666.000
save_config: set [delta_calibrate] height3 = 0.0
save_config: set [delta_calibrate] height3_pos = 28402.000,21643.000,28643.000
save_config: set [delta_calibrate] height4 = 0.0
save_config: set [delta_calibrate] height4_pos = 29683.000,23529.000,23592.000
save_config: set [delta_calibrate] height5 = 0.0
save_config: set [delta_calibrate] height5_pos = 27694.000,27784.000,21817.000
save_config: set [delta_calibrate] height6 = 0.0
save_config: set [delta_calibrate] height6_pos = 23501.000,31140.000,23673.000
```

It's not clear to me why you always get a failure in the same spot.  I'll put together a fix, but it will take me some time to do that.  In the mean time, manually applying the basic calibration results should work.

@wakeboy180 - if you are experience a problem please attach the klipper log file from the event as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

DanilaX1, 2018-10-09:
Kevin, I'm glad for your help. I did the calibration again. saving configuration and command m112. I attach the log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2461503/klippy.log)

KevinOConnor, 2018-10-15:
I believe this issue should now be fixed (commit 368703f).

-Kevin

DanilaX1, 2018-10-17:
Thanks now works perfectly! Thanks for your work. And such a good product.

closed 2018-10-22 by KevinOConnor