Klipper3d/klipper #734: OrangePi crash

povlhp, 2018-10-03:
Nothing useable in log.
Just tried to print a turbo speed (accel 4500, max speed 250, travel 500 in gcode), and my OrangePi lite (quad core, 1200MHz) lost wireless networking (did not come up again), printer froze for like a minute, and then started again. A few layers later, at hang again, and I rebooted my OrangePi.

Nothing special in any logfiles, not in klipper.log, syslog, messages etc. So if anything was written, it surely wasn't flushed.

povlhp, 2018-10-03:
Had another crash on a print. This time printing just stopped. No dead Pi.
Log attached.
This is likely the relevant lines:, as Octoprint stood with must home first.

The jump in 5mm is because that is my abort action (set relative coordinates and move Z 5mm up).
This is with firmware just installed from GitHub an hour ago. Assumed the new serial code with no interrupts maybe could have solved my first problem.

```
Stats 5615.6: gcodein=1510459 mcu: mcu_awake=0.070 mcu_task_avg=0.000206 mcu_task_stddev=0.000223 bytes_write=3687863 by
tes_read=771221 bytes_retransmit=97 bytes_invalid=0 send_seq=73022 receive_seq=73022 retransmit_seq=72953 srtt=0.011 rtt
var=0.005 rto=0.032 ready_bytes=0 stalled_bytes=0 freq=15999941 heater_bed: target=65 temp=65.1 pwm=0.252 print_time=374
8.968 buffer_time=2.215 print_stall=0 extruder: target=205 temp=205.2 pwm=0.456
Must home axis first: 69.598 112.951 0.220 [3004.133]
Stats 5616.6: gcodein=1510616 mcu: mcu_awake=0.083 mcu_task_avg=0.000191 mcu_task_stddev=0.000206 bytes_write=3688505 bytes_read=771541 bytes_retransmit=97 bytes_invalid=0 send_seq=73042 receive_seq=73041 retransmit_seq=72953 srtt=0.013 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=16 freq=15999941 heater_bed: target=65 temp=65.0 pwm=0.252 print_time=3749.688 buffer_time=1.934 print_stall=0 extruder: target=205 temp=205.2 pwm=0.449
Must home axis first: 90.402 112.951 5.220 [3003.475]
Must home axis first: 90.402 112.951 5.220 [3003.475]
Stats 5617.6: gcodein=1510979 mcu: mcu_awake=0.083 mcu_task_avg=0.000191 mcu_task_stddev=0.000206 bytes_write=3688725 bytes_read=771775 bytes_retransmit=97 bytes_invalid=0 send_seq=73055 receive_seq=73055 retransmit_seq=72953 srtt=0.014 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=16 freq=15999941 heater_bed: target=0 temp=65.0 pwm=0.000 print_time=3750.088 buffer_time=1.333 print_stall=0 extruder: target=0 temp=204.8 pwm=0.000

```

Here the gcode from console just before the error:
```
Send: N2846 G1 X90.002 Y111.751 E265.8425*89
Recv: ok
Send: N2847 G0 F1620 X90.002 Y112.551*74
Recv: ok
Send: N2848 G0 F2400 X90.402 Y112.951*78
Recv: ok
Send: N2849 G1 F3000 E265.86778*54
Recv: ok
Send: N2850 G1 F1800 X69.598 Y112.951 E266.52513*4
Recv: !! Must home axis first: 69.598 112.951 0.220 [3004.133]
Changing monitoring state from "Printing" to "Cancelling"
Send: N2851 G91*47
Recv: ok
Send: N2852 M117 1% L=0/59*111
Recv: ok
```

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/2443690/klippy.log.gz)

KevinOConnor, 2018-10-05:
The "Must home axis first" error occurs when a movement occurs prior to homing a particular axis and/or  trying to move an axis after turning off the motors.  The log didn't go back far enough to include the list of commands Klipper received prior to the first failure, so it's hard to say what in particular commanded the motors off.

-Kevin

KevinOConnor, 2018-10-08:
None of those things would cause the software to report a "must home axis" error - only the two situations I listed above should lead to that error.

-Kevin

KevinOConnor, 2018-10-16:
Closing for now.  Let us know if it reappears and you can't track it down.

closed 2018-10-16 by KevinOConnor