Klipper3d/klipper #736: Very long pauses while printing

lar3ry, 2018-10-04:
While printing, I am getting very long pauses, anywhere from .5 seconds to about 3 seconds. I had previously printed a cylinder of 10mm diameter, 20mm high, with a 4mm hole up the Z axis.
This was pretty good. better than I had with Marlin. Tried a print with larger circles (about 45mm), which showed the problem. I had thought it was pausing only on the circles, but in further testing, I see that it's pausing during infill (25% honeycomb) as well. Speed of perimeters was 80mm/sec, a speed at which I often printed using Marlin. 

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2444982/klippy.zip)

jakep82, 2018-10-04:
https://github.com/KevinOConnor/klipper/issues/513

lar3ry, 2018-10-04:
I had a look at that. 
Not sure what to do about it.
Further info: Sliced with Slic3rPE. I did a load graph and attached it. The first run was showing the problem, and the second one I stopped with an M112 after I counted 5 pauses.
I THOUGHT I was printing with the virtual SD card, but I'm really not sure.
The file I ws using was the file I first noticed the problem on, but modfied to remove all extruder moves and temperature settings. 
Is there any other info you want from me, or is there something I can do to prevent this?

[image]

jakep82, 2018-10-04:
The load graph makes it quite clear you weren't printing from the virtual SD. If you were, your host buffer would be zero. Just enabling this feature in your printer.cfg is not enough. You also need to initialize the SD in Octoprint and select the correct version of the file. You can change the file view in Octoprint so it only shows files on the SD. You will also need to manually update the file list any time you upload something new.

lar3ry, 2018-10-04:
OK... well, here's what I did...
1. enabled virtual_sdcard pointed at ~/.octoprint/uploads/
2. restarted Klipper
3. clicked only show SD files. No files showed.
4. selected initialize SD card. This caused most of my files to show up. The ones missing were in a directory
5. still set on Only show SD files, I selected a file did a Preheat, and clicked Print.

I may not have refreshed the SD card on subsequent attempts, if that's what you mean by "update the file list". I was uploading new versions of the file, modified to not extrude or heat.

I just went though those steps again, and am going to start the print that did a lot of pausing.

jakep82, 2018-10-04:
When it's printing, take note of the file name in the State section in Octoprint.  If it's printing from the virtual SD, it will show the file name with (SD) appended to the end.  If it doesn't have the SD, then you're not printing a virtual SD file.

lar3ry, 2018-10-04:
Another good tip! Thanks again!

closed 2018-10-04 by lar3ry