Klipper3d/klipper #737: Issue Configuring dual-z and bltouch

tifrei, 2018-10-04:
Getting following report: 
Recv: // Klipper state: Not readyRecv: // pin ar30 used multiple times in configRecv: // Once the underlying issue is corrected, use the "RESTART"Recv: // command to reload the config and restart the host software.Recv: // Printer is haltedRecv: !! pin ar30 used multiple times in config

used following code for setting up MKS GEN 1.4, bltouch and Dual-z stepper driver

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
position_endstop: 0
endstop_pin: z_virtual_endstop
position_max: 430

[stepper_z1]
step_pin: ar36
dir_pin: !ar34
enable_pin: !ar30
step_distance: .0025

# Define the BLTouch servo
[servo bltouch]
pin: ar32
maximum_servo_angle: 180
minimum_pulse_width: 0.0006
maximum_pulse_width: 0.0024

# Define a probe using the BLTouch
[probe]
pin: ar30
z_offset: 0.2
activate_gcode:
    SET_SERVO SERVO=bltouch ANGLE=10
    SET_SERVO SERVO=bltouch ANGLE=60
    G4 P200
deactivate_gcode:
    SET_SERVO SERVO=bltouch ANGLE=90
    G4 P100

# Example bed_tilt config section
[bed_tilt]
points:
    100,100
    10,10
    10,100
    10,190
    100,10
    100,190
    190,10
    190,100
    190,190

# If the BLTouch is used to home the Z axis, then define a
# homing_override section, use probe:z_virtual_endstop as the
# endstop_pin in the stepper_z section, and do not set
# position_endstop in the stepper_z section.

[homing_override]
set_position_z: 5
axes: z
gcode:
    ; G90   ; Uncomment these 2 lines to blindly lift the Z 2mm at start
    ; G1 Z7 F600
    G28 X0 Y0
    G1 X100 Y100 F3600
    G28 Z0

tifrei, 2018-10-04:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2446727/klippy.log)

tifrei, 2018-10-04:
Any idea of what pin has to be used for either the z1 or bltouch if not the ar30?

theopensourcerer, 2018-10-04:
No sorry. I've not used a Bltouch. 

The problem is you have pin ar30 configured for your z1 stepper "enable" as well as your probe... It can't be both ;-)

KevinOConnor, 2018-10-11:
It looks like the original question was answered (don't use the same pin twice in the config file).

closed 2018-10-11 by KevinOConnor