Klipper3d/klipper #744: Option 'position_endstop' is not valid in section 'stepper_z'

bruce356, 2018-10-05:
Hi, after updating to the latest version received this error:-
Option 'position_endstop' is not valid in section 'stepper_z'
This entry in printer.cfg was never a problem previously.
What should be done just remove the entire line or change the value?

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2448883/klippy.zip)
Regards - bruce

closed 2018-10-05 by bruce356

AmyTheCute, 2021-01-04:
how did you fix it?

MrEd88, 2021-01-25:
Just updated and found the same issue. Check this link:

https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md#changes

"20201213: It is not valid to specify a Z "position_endstop" when using "probe:z_virtual_endstop". An error will now be raised if a Z "position_endstop" is specified with "probe:z_virtual_endstop". Remove the Z "position_endstop" definition to fix the error."

I am reading the link to find any other issues I might find.