Klipper3d/klipper #745: Other language support for menu

ich777, 2018-10-05:
@mcmatrix is it possible to change the language if yes, how? Or do i have to make my complete own custom menu?

mcmatrix, 2018-10-05:
At the moment there's no support for the language.
You have to override the whole menu to make it appear in the different language.
Don't make changes in `menu.cfg`!
If you use the config name of the same menu item, you can add menu changes to your printer config file. Your changes will redefine an existing menu item.

KevinOConnor, 2018-10-25:
It doesn't look like this is being worked on right now, so I'm going to close it.  I'll reopen if someone starts actively working on it.

closed 2018-10-25 by KevinOConnor

manu7irl, 2018-11-02:
@mcmatrix 
Hi can you explain a bit more on restructuring the menu? 
because I want to make it work in rtl language hebrew or arabic.
can it be done?